MikePopoloski/slang #467: Adding a lint warning for code typo that creates infinite loop

jrudess, 2021-09-30:
The following while statement has an extraneous semi-colon which creates an infinite-loop.   The Big Three accept the code without warning since it is legal syntax.  

But an infinite-loop gets created if the loop-expression is true for both while and do-while loops, would a warning make sense for empty statements; e.g. a bare semi-colon or an empty begin/end pair.

```systemverilog
module top;
    bit done;
    function automatic is_done();
        return 1'b1;
    endfunction
    initial begin
        done = 0;
        while (done == 0);
        begin
            done = is_done();
        end
    end
endmodule
```

MikePopoloski, 2021-09-30:
Yes, in fact I have an internal list of warnings I want to add and this is already on it (might be worth putting in an issue somewhere). There is already a Wempty-stmt warning but it only applies to empty statements in a block; it should probably be a separate warning for single-statement cases for loop / conditional bodies.

MikePopoloski, 2022-06-26:
This has been added:
```
source.sv:8:26: warning: while loop has empty body (put the semicolon on a separate line to silence this warning) [-Wempty-body]
        while (done == 0);
                         ^
```

closed 2022-06-26 by MikePopoloski