MikePopoloski/slang #468: [Question or Feature Request] Is there any way to build slang so that `slang --version` prints the git commit id?

chengniansun, 2021-10-01:
Thanks.

jrudess, 2021-10-01:
If you clone the git-repo and build from there instead of the tarball, it'll include the commit hash.  I'm guessing this is more about whether the nightly release should include the git-hash in the version also?

chengniansun, 2021-10-02:
@jrudess Thanks for your reply. 

No wonder. I have been using the nightly release and built slang from the nightly tarball.

chengniansun, 2021-10-02:
Closing this issue, and I will use the git to checkout the source code instead. Thanks.

MikePopoloski, 2021-10-02:
Huh, I was not aware that the CI build wasn't getting the git revision in the version string. I don't see why that build would be any different; I can investigate further.

MikePopoloski, 2021-10-02:
Oh, I think I misunderstood. The prebuilt binaries do indeed have the git revision in them, but if you just grab a source copy tarball and build yourself there will be no git hash because the only way to know it is to be inside a git repository (the build runs git rev-parse).

closed 2021-10-02 by MikePopoloski

chengniansun, 2021-10-03:
You are right. I used the source tarball. I am not building nightly binary from the github now.