MikePopoloski/slang #482: rand_mode and task vs function

jrudess, 2021-10-16:
It's quite unfortunate that the LRM defines rand_mode(bit on_off) as a task and not a function.  I did some testing with VCS and Questa, and both allow calling the rand_mode() "task" from a function without warning or error.

In the example, a parse error is given if rand_mode() is called on a class object from within a function, but calling rand_mode is allowed on the class random variables.  Can an LRM extension be added to also allow calling rand_mode on the object?

```systemverilog
slangtest48.sv:4:9: error: cannot invoke a task from within a function
        rand_mode(0);
        ^~~~~~~~~~~~
slangtest48.sv:5:9: error: cannot invoke a task from within a function
        this.rand_mode(0);
        ^~~~~~~~~~~~~~
slangtest48.sv:13:9: error: cannot invoke a task from within a function
        c.rand_mode(0);
        ^~~~~~~~~~~
```
```systemverilog
class C;
    rand int i;
    function void f();
        rand_mode(0);
        this.rand_mode(0);
        i.rand_mode(0);
    endfunction
endclass
module top;
    C c;
    function make();
        c = new();
        c.rand_mode(0);
    endfunction
    initial begin
        make();
    end
endmodule
```

jrudess, 2021-10-16:
I think constraint_mode should be the same.   I did a quick trial and all three simulators allow constraint_mode(0) from a function.  I've never actually run into object.constraint_mode(0) in the wild, only [object.]constraint.constraint_mode(0);

MikePopoloski, 2021-10-17:
Done in 19c836a52438474083a188a395c627c3cd404253

closed 2021-10-17 by MikePopoloski