MikePopoloski/slang #483: Option to relax 'lifetime specifier is not allowed on prototype declaration' from error to warning

jrudess, 2021-10-16:
What do you think about either adding a command-line option or permanently relaxing the below error to be a warning and to drop/ignore the automatic specifier in prototype declarations?  I'm mostly running into this in code-libraries from the 2008-2011 time frame, when folks were still confused about whether class methods were automatic by default.

```systemverilog
slangtest49.sv:2:17: error: lifetime specifier is not allowed on prototype declaration
    extern task automatic t();
                ^~~~~~~~~
```
```systemverilog
class C;
    extern task automatic t();
endclass
task automatic C::t();
endtask
```

jrudess, 2021-10-16:
Was playing around with the static keyword, and think there may be a missed error case in this example.   LRM $8.10 says that the static lifetime specifier is illegal for tasks inside a class, but the final implementation of C::t2() doesn't flag an error with the below test case.

```systemverilog
class C;
    extern task t1();
    extern task t2();
endclass
task automatic C::t1();
endtask
task static C::t2();
endtask
```

MikePopoloski, 2021-10-17:
Yeah, relaxing to a warning seems fine to me. Done in 5ffb1d6a7892a9f114b36ff9838922bd08e51915

You're right, I was missing the static lifetime check for out-of-block methods. Fixed in 49ae51746f787b71f97065ec9fc16eac2f803dd0

closed 2021-10-17 by MikePopoloski