MikePopoloski/slang #493: Typo scenario in for loop 3rd argument (for_step)

jrudess, 2021-10-29:
I've added Slang into my build pipe now, so I can collect cases where Slang is passing but other simulators are failing.   In this case VCS flags an error at the ')' for the for-statement as being unexpected since the 3rd argument is just a variable.

Looking at the LRM grammar in 12.7, I don't think the bare-variable matches `for_step_assignment` -- should this be an error?
```
    loop_statement ::== ( [ for_initialization ] ; [ expression ] ; [ for_step ] )
    for_step ::== for_step_assignment { , for_step_assignment }
    for_step_assignment ::== operator_assignment | inc_or_dec_expression | function_subroutine_call
```
```systemverilog
module top;
    initial begin
        for (int i = 0; i < 10; i) begin
        end
    end
endmodule
```

MikePopoloski, 2021-10-31:
Yes, I agree this should be an error. This is a case where the LRM is being overly restrictive for no good reason I think, but no reason not to enforce the rule if the other simulators do too.

jrudess, 2021-10-31:
For completeness, confirmed the others also flag it.

Questa
```
'i' is not a task name
```
Xcelium
```
        for (int i = 0; i < 10; i) begin
                                 |
expecting an '=' or '<=' sign in an assignment [9.2(IEEEE)]
        for (int i = 0; i < 10; i) begin
                                       |
expecting a non-blocking assignment in this enhanced for loop initializer or incrementer specification.
        for (int i = 0; i < 10; i) begin
                                       |
illegal lvalue syntax [9.6(IEEE)]
```

MikePopoloski, 2021-11-07:
Fixed in 98a85f3ade12db507377041d40071c453b37e7f9

closed 2021-11-07 by MikePopoloski