MikePopoloski/slang #495: Undefined reference when using slang as a library

Kuree, 2021-11-13:
Error message:
```
/usr/bin/ld: CMakeFiles/test.dir/test.cc.o: in function `main':
test.cc:(.text.startup+0x39): undefined reference to `slang::DiagnosticEngine::reportAll[abi:cxx11](slang::SourceManager const&, std::span<slang::Diagnostic const, 18446744073709551615ul>)'
collect2: error: ld returned 1 exit status
```

How to reproduce:
```
mkdir test && cd test
git clone https://github.com/MikePopoloski/slang/
```
Add these two files in the `test` folder

- `CMakeLists.txt`
```cmake
cmake_minimum_required(VERSION 3.12)
project(test)

set(CMAKE_CXX_STANDARD 20)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)

# external library
add_subdirectory(slang)
add_executable(test test.cc)
target_link_libraries(test PRIVATE slangcompiler)
```
- `test.cc`
```C++
#include "slang/text/SourceManager.h"
#include "slang/diagnostics/DiagnosticEngine.h"

int main() {
    slang::SourceManager s;
    slang::DiagnosticEngine::reportAll(s, {});
}
```

Then run
```
mkdir build && cd build && cmake ..
make test -j
```

Tested with `gcc-11` and `clang-12`.

Is there any way to resolve this issue? `driver` seems to use that function and it is linked just fine using this example.

Thanks.

MikePopoloski, 2021-11-14:
You need to link the slangcore lib as well, there isn't just one library. slangcompiler should have a cmake dependency on the slangcore lib, but I guess it's not getting pulled in for some reason?

Kuree, 2021-11-14:
Still the same error after linking `slangcore`.
```cmake
target_link_libraries(test PRIVATE slangcore slangcompiler)
```

MikePopoloski, 2021-11-14:
Are you sure it's actually linking the libraries you're specifying? That's where the function is defined, so I'm not seeing what the problem is.

Kuree, 2021-11-14:
I'm quite certain? Here is the verbose make output:
```
/usr/bin/cmake -E cmake_link_script CMakeFiles/test.dir/link.txt --verbose=1
/usr/bin/clang++-12 -O3 -DNDEBUG CMakeFiles/test.dir/test.cc.o -o test  slang/lib/libslangcompiler.a slang/lib/libslangcore.a slang/lib/libslangparser.a slang/lib/libslangcore.a -lstdc++fs 
/usr/bin/ld: CMakeFiles/test.dir/test.cc.o: in function `main':
test.cc:(.text+0x22): undefined reference to `slang::DiagnosticEngine::reportAll[abi:cxx11](slang::SourceManager const&, std::span<slang::Diagnostic const, 18446744073709551615ul>)'
```

MikePopoloski, 2021-11-14:
I followed your exact steps listed originally with a fresh checkout and it compiled just fine for me. Not sure what's going on...

Kuree, 2021-11-14:
That's odd. Here is how to reproduce it using Docker. I'm on the latest Ubuntu so the toolchain could be the issue.

```Dockerfile
FROM ubuntu:21.10

RUN apt update
ENV DEBIAN_FRONTEND="noninteractive"
RUN apt install -y build-essential python3 python3-pip cmake git vim
```

Here is the tools installed via `build-essential`:
- `ld`: 2.37
- `gcc`: 11.2.0

MikePopoloski, 2021-11-14:
Hmm, I'm on Ubuntu 20.04, and was using gcc 9.3. I can try with with docker though.

Kuree, 2021-11-14:
I can confirm that `gcc-9` works on Ubuntu 21.10. So this is definitely compiler-related issue.

Kuree, 2021-11-14:
Found a temporary fix:

```diff
diff --git a/external/span.hpp b/external/span.hpp
index 97b52194..df11cb3b 100644
--- a/external/span.hpp
+++ b/external/span.hpp
@@ -197,7 +197,7 @@
 # define  span_HAVE_STD_SPAN  0
 #endif
 
-#define  span_USES_STD_SPAN  ( (span_CONFIG_SELECT_SPAN == span_SPAN_STD) || ((span_CONFIG_SELECT_SPAN == span_SPAN_DEFAULT) && span_HAVE_STD_SPAN) )
+#define  span_USES_STD_SPAN 0 // ( (span_CONFIG_SELECT_SPAN == span_SPAN_STD) || ((span_CONFIG_SELECT_SPAN == span_SPAN_DEFAULT) && span_HAVE_STD_SPAN) )
 
 //
 // Use C++20 std::span:
```

It seems like these new compilers enables native C++20 `std::span` implementation, which doesn't work out of the box with slang.

MikePopoloski, 2021-11-14:
Oh, it's building with a different C++ version perhaps? If the defines resolve differently when compiling the slang libs versus the test.cc file where you include a header then you'll get definition mismatches, which would show up as a linker error.

MikePopoloski, 2021-11-14:
When statically linking you want to make sure all build options are the same.

Kuree, 2021-11-14:
I think it might because slang sets CXX standard to 17 so it uses span-lite, when I included it as a sub-directory, the compiler switches to `std::span` for my `test.cc`.

Kuree, 2021-11-14:
The fix is to add the following definition to the target to force non-std version of span. This is only required if the target is built against C++20.

```cmake
target_compile_definitions(test PUBLIC span_CONFIG_SELECT_SPAN=1)
```

Thanks for the help!

closed 2021-11-14 by Kuree

MikePopoloski, 2021-11-14:
Glad you figured it out. I'll look into making slang easier to pull into cmake projects in the future.