MikePopoloski/slang #497: preprocessor crashes on debug build

mytbk, 2021-11-15:
I write a preprocessor program and run it with ``./preprocess -I test/ test/top.v``. It gets a double free or corruption error.

Backtrace:

```
#0  0x00007ffff7a92d22 in raise () from /usr/lib/libc.so.6
#1  0x00007ffff7a7c862 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff7ad4d28 in __libc_message () from /usr/lib/libc.so.6
#3  0x00007ffff7adc92a in malloc_printerr () from /usr/lib/libc.so.6
#4  0x00007ffff7ade1b0 in _int_free () from /usr/lib/libc.so.6
#5  0x00007ffff7ae19e8 in free () from /usr/lib/libc.so.6
#6  0x000055555555f577 in slang::SmallVector<slang::logic_t>::cleanup (this=0x7fffffffd570)
    at /home/irudog/verilog-slang/include/slang/util/SmallVector.h:269
#7  slang::SmallVector<slang::logic_t>::~SmallVector (this=0x7fffffffd570, __in_chrg=<optimized out>)
    at /home/irudog/verilog-slang/include/slang/util/SmallVector.h:212
#8  slang::SmallVectorSized<slang::logic_t, 16ul>::~SmallVectorSized (this=0x7fffffffd570, __in_chrg=<optimized out>)
    at /home/irudog/verilog-slang/include/slang/util/SmallVector.h:276
#9  slang::NumberParser::~NumberParser (this=0x7fffffffd538, __in_chrg=<optimized out>)
    at /home/irudog/verilog-slang/include/slang/parsing/NumberParser.h:19
#10 slang::Preprocessor::~Preprocessor (this=0x7fffffffd140, __in_chrg=<optimized out>)
    at /home/irudog/verilog-slang/include/slang/parsing/Preprocessor.h:55
#11 0x0000555555560e05 in slang::Preprocessor::predefine (this=0x7fffffffdb10, definition="__slang__ 1", fileName="<api>")
    at /home/irudog/SourceCode/slang/source/parsing/Preprocessor.cpp:67
#12 0x00005555555610c4 in slang::Preprocessor::undefineAll (this=0x7fffffffdb10)
    at /home/irudog/SourceCode/slang/source/parsing/Preprocessor.cpp:83
#13 0x00005555555606a9 in slang::Preprocessor::Preprocessor (this=0x7fffffffdb10, sourceManager=..., alloc=..., diagnostics=..., options_=...)
    at /home/irudog/SourceCode/slang/source/parsing/Preprocessor.cpp:28
#14 0x000055555555dd51 in main (argc=4, argv=0x7fffffffe0e8) at preprocess.cc:40
```

My code:

```
#include <slang/parsing/Preprocessor.h>
#include <slang/util/BumpAllocator.h>
#include <slang/diagnostics/Diagnostics.h>
#include <slang/text/SourceManager.h>
#include <iostream>

void usage(const char *p)
{
  std::cerr << p << " [-I <path>]... <file>" << std::endl;
  exit(1);
}

int main(int argc, char *argv[])
{
  slang::BumpAllocator alloc;
  slang::Diagnostics diag;
  slang::SourceManager srcMgr;

  int pargc = 1;
  std::string filename;
  while (pargc < argc) {
    if (strcmp(argv[pargc], "-I") == 0) {
      pargc++;
      if (pargc < argc) {
	srcMgr.addUserDirectory(argv[pargc]);
      } else {
	usage(argv[0]);
      }
    } else if (filename.size() == 0) {
      filename = argv[pargc];
    } else {
      usage(argv[0]);
    }
    pargc++;
  }
  if (filename.size() == 0) {
    usage(argv[0]);
  }

  slang::Preprocessor preprocessor(srcMgr, alloc, diag);
  preprocessor.pushSource(srcMgr.readSource(filename));

  std::string result;
  while (true) {
    slang::Token token = preprocessor.next();
    result += token.toString();
    if (token.kind == slang::TokenKind::EndOfFile)
      break;
  }

  std::cout << result;
}
```

mytbk, 2021-11-15:
However the slang driver doesn't crash, I don't know what the difference is.

MikePopoloski, 2021-11-16:
There's nothing obviously wrong with your code; it makes me suspect some ABI mismatch between the slang lib and the source file you're compiling. Perhaps mismatched compilation flags or something?

mytbk, 2021-11-16:
I tried again on a cleanly installed Debian system with GCC 10.2.1, still crashes.
I just use ``cmake -GNinja -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX="$HOME/verilog-slang" ..`` to build slang, and use ``g++ -g -Og -std=c++17 -I ~/verilog-slang/include/ -c preprocessor.cc`` to compile the program, and use ``g++ -o preprocessor preprocessor.o -L ~/verilog-slang/lib/ -lslangparser -lslangcore`` to link.

I've tried using valgrind and address sanitizer to check the program, from the valgrind log ``data_`` in SmallVector may be uninitialized (but I don't know why), and ASAN just says bad-FREE and gives no useful information.

I also thought the move constructor of SmallVectorSize may have problem and use this patch, but it seems unrelated and doesn't solve the problem:

```
diff --git a/include/slang/util/SmallVector.h b/include/slang/util/SmallVector.h
index 5b4b1507..2bd7f7a8 100644
--- a/include/slang/util/SmallVector.h
+++ b/include/slang/util/SmallVector.h
@@ -300,7 +300,7 @@ public:
             this->len = other.len;
             this->capacity = other.capacity;
 
-            other.data_ = nullptr;
+            other.data_ = reinterpret_cast<T*>(&other.firstElement[0]);
             other.len = 0;
             other.capacity = 0;
         }
```

mytbk, 2021-11-16:
I know what happened now. It is an ABI mismatch.
When using CMAKE_BUILD_TYPE=Debug, slang is built with ``-DDEBUG`` compiler flags, which makes ``class SourceLocation`` bigger because it has an additional ``string_view bufferName``, thus makes ``class NumberParser`` bigger. If the apps that uses them does not use ``-DDEBUG``, the size of the classes does not match the ones in the built object files, thus causes this strange issue.
Using ``-DDEBUG`` to build my app solves this issue.

closed 2021-11-16 by mytbk

MikePopoloski, 2021-11-16:
Great, glad you figured it out!