MikePopoloski/slang #498: Parse error for some expressions

mytbk, 2021-11-19:
I build a syntax tree and print it out again with the following:

```
#include <slang/syntax/SyntaxTree.h>
#include <slang/syntax/SyntaxPrinter.h>

#include <iostream>

int main(int argc, char *argv[])
{
  if (argc != 2) {
    std::cerr << "usage: " << argv[0] << " file" << std::endl;
    return 1;
  }

  auto tree = slang::SyntaxTree::fromFile(argv[1]);

  std::cout << slang::SyntaxPrinter::printFile(*tree) << std::endl;
}
```

When I have a file with ``foo(32'h0e+32'h4000)``, it emits ``foo(32'h0e+32+32'h'h4000)`` which is illegal sv code.

Also when I use the slang driver to test this code, which both iverilog and verilator can run:

```
task foo(bit [31:0] a);
	$display("%x", a);
endtask: foo

task test();
	foo(32'h0e+32'h4000);
endtask: test

module m;
initial begin
	test();
end
endmodule
```

slang reports:

```
test/test.sv:6:2: error: too many arguments for 'foo'; expected 1 but 2 were provided
	foo(32'h0e+32'h4000);
	^~~~~~~~~~~~~~~~~~~~
test/test.sv:6:15: error: expected vector literal digits
	foo(32'h0e+32'h4000);
	             ^
```

When I add spaces around the '+' operator, these errors disappears.

MikePopoloski, 2021-11-21:
Thanks for the report, fixed in f4898d82d0649d2d248c75ce85ae32eab063c64b

closed 2021-11-21 by MikePopoloski

mytbk, 2021-11-22:
This bug is not totally solved. Now slang can parse this, but the printing of the tree is still wrong, ``foo(32'h0e+32'h4000)`` is now printed as ``foo(32'h0e+32+32'h4000)``.