MikePopoloski/slang #504: build succeeds, but using --ast-json gives internal error?

Practical-UVM-Step-By-Step, 2021-11-30:
HI:
Thank you for writing this parser!
I was running a few more testcases from my work against your code. That code is also on Github BTW.
I have included a basic UVM example, and tried to run slang against it. 
The Build succeeds without the --ast-json switch.

 But when I used the --ast-json <filename> option, it gives me an Error on Expression.cpp line 63 saying the default case should be unreachable.  That seems like an issue in the code.

Attached is a testcase. It's a bit big as I made it self contained, and its one of my smaller examples. There is a runme in the tarball.  Please change the path to path of your slang binary and you can see the error.

[testcase.tar.gz](https://github.com/MikePopoloski/slang/files/7623653/testcase.tar.gz)

A request:  is it possible to use the library without Elaboration in a program? IE: If I use SLANG as a parser ONLY and get a raw syntax Tree that I have to walk through?
if so, can you point me to an example in your tests that I can use as a reference? 

Thank you very much.

MikePopoloski, 2021-11-30:
Thanks for the report, I'll try to reproduce from the testcase.

Yes, it's definitely possible to use slang as a library for parsing only. The rewriter tool in the repo has a minimal example of parsing and writing the tree back to file. You can walk the resulting tree manually or make use of the SyntaxVisitor helper class.

MikePopoloski, 2021-12-11:
Fixed in 269293c5f0cedd50b76c1e70de0e77ea0eae3c5b

closed 2021-12-11 by MikePopoloski