MikePopoloski/slang #507: Add better path support for include files

DeeeeLAN, 2022-01-01:
Copied from our conversation:

> There are probably a few decent options. One option would be to just resolve every relative path so it using the full system path, then every path would be standardized. This would probably be the simplest option to go with. Another option would be to provide some sort of access to the map so I can either manually override the buffer a path points to, or at least see what paths are used for the keys and I could copy the key into the assignText call to make sure it matches. But overriding the buffer directly is probably a bad idea at the moment since there is no support for changing the comp unit in this matter. I imagine that will be a bigger task but that would be nice to have eventually, being able to take a source manager or compilation unit, and either changing a file or creating a new sm/comp with one or more files swapped out or deleted.

MikePopoloski, 2022-01-26:
I reworked how include path handling is done for in-memory text buffers. I think it should probably work as you expect now.

closed 2022-01-26 by MikePopoloski

DeeeeLAN, 2022-01-26:
Awesome, I will give it a try and see if it does the job I am hoping it will do. Not sure when I will get to it.