MikePopoloski/slang #509: Regression for $past usage in an assert-property with recent const changes

jrudess, 2022-01-17:
Starting with 4e8cb3c49ef160a6dedec943087dc7ab71eedd91 I have a regression for just one assert-property in a medium sized project.  I haven't yet been able to reproduce a failure with a minimized example, but will update when reproduced.  In the meantime, here's the original error (obfuscated).  The printed code is the consequent expression of an implication operator for the failing assert-property.

```systemverilog
source.sv:2928:76: error: '$past' is not allowed in a constant context
              ~`CHECK_VALUE($past(arr1[g_i][g_j].value), $past(arr2[g_i][g_j]), ~disable) & arr3[g_i],
                                                         ^~~~~~~~~~~~~~~~~~~~~
```

Here's the define breakout:
```systemverilog
`define ZX(in, sz) {{sz-$bits(in){1'b0}}, in}

`define CALC_VALUE(b, c) \
    (~(PARAM_VAL1 && c)) ? '0 : \
        (PARAM_VAL2 == 2)
            ? {^{`ZX(b,500) & {250{2'b10}}}, ^{`ZX(b,500} & {250{2'b01}}}} \
            : {^{b}} ))

`define CHECK_VALUE(a, b, c) ((a == `CALC_VALUE(b, c)) || ~(PARAM_VAL1 && c))
```

My suspicion is that passing the result from $past into $bits is triggering the problem.

jrudess, 2022-01-17:
Minimized example:

```systemverilog
module top;
    logic clk, reset, a, b, c;
    assert property(@(posedge clk) disable iff (reset)
        a |-> {500-$bits($past(b)){^{c}}});
endmodule
```
```systemverilog
slangtest60.sv:4:26: error: '$past' is not allowed in a constant context
        a |-> {500-$bits($past(b)){^{c}}});
                         ^~~~~~~~
```

This may not be a Slang bug, but the original code compiles with the big-three.   It would be possible to refactor the assertion to use $bits(b) instead of $bits($past(b)).

MikePopoloski, 2022-01-18:
Thanks for the minimized reproducer. This does indeed look wrong, the expression in $bits() shouldn't need to be constant.

MikePopoloski, 2022-01-20:
Fixed in 893f7712ef1a6824d0635f4b1f708000087406d4

closed 2022-01-20 by MikePopoloski