MikePopoloski/slang #512: Missing semi-colon on typedef variable declaration leads to confusing error if typedef is imported from package

jrudess, 2022-02-09:
Go ahead and close this one if you don't think there is anything reasonable to do given the syntax involved in the scenario.  

One of my co-workers hit a confusing error that was the result of a missing semicolon (line 11, declaration of 'a').  I was only able to reproduce the scenario when I placed the typedef in the package.

```systemverilog
package def_pkg;
    typedef logic node;
endpackage

module M
    import def_pkg::*;
(
    input node clk,
    input node rst
);
    node a
    node b;
endmodule
```
```systemverilog
slangtest64.sv:8:11: error: import of 'node' collides with an existing declaration
    input node clk,
          ^~~~
slangtest64.sv:12:5: note: declared here
    node b;
    ^
slangtest64.sv:6:21: note: imported from here
    import def_pkg::*;
                    ^
slangtest64.sv:2:19: note: declared here
    typedef logic node;
                  ^
slangtest64.sv:9:11: error: import of 'node' collides with an existing declaration
    input node rst
          ^~~~
slangtest64.sv:12:5: note: declared here
    node b;
    ^
slangtest64.sv:6:21: note: imported from here
    import def_pkg::*;
                    ^
slangtest64.sv:2:19: note: declared here
    typedef logic node;
                  ^
slangtest64.sv:11:5: error: import of 'node' collides with an existing declaration
    node a
    ^~~~
slangtest64.sv:12:5: note: declared here
    node b;
    ^
slangtest64.sv:6:21: note: imported from here
    import def_pkg::*;
                    ^
slangtest64.sv:2:19: note: declared here
    typedef logic node;
                  ^
slangtest64.sv:11:5: error: 'node' is not a type
    node a
    ^~~~
slangtest64.sv:11:11: error: expected ','
    node a
          ^
slangtest64.sv:12:9: error: expected ','
    node b;
        ^
```

jrudess, 2022-02-10:
The final three errors mention the commas, and I think  those three errors are close enough to the syntax point that the typo becomes obvious.   When the typedef isn't in the package, those are the only three errors displayed.

The only reason it becomes confusing when the typedef is in the package is that the very first error repeats for every signal in the port list (which usually is a long list), so the more helpful errors (which are related to the guess of missing commas) are pushed way down the error-chain.

MikePopoloski, 2022-02-12:
I made some improvements in 513b191b9a831a5da277a56363434d1be99363aa that I think will help.

closed 2022-02-12 by MikePopoloski