MikePopoloski/slang #513: Detection of multiple driver scenarios during elaboration

jrudess, 2022-02-16:
I'm starting to see some broader uptake in build flows with Slang as a 1st stage compiler due to its speed.  One of the most common coding mistakes getting past the Slang stage involves multiple-driver errors.  I thought I had previously seen an issue tracking this implementation, but didn't see one currently open.

```systemverilog
module top;
    logic a;
    logic clk;
    logic rst;

    always_ff (@posedge clk) begin
        if (rst) a <= '0;
        else     <= '1;
    end

    always_comb begin
        a = 1;
    end

endmodule
```systemverilog

VCS will then report an error like:
```
Error-[ICPD] Illegal combination of drivers
   slangtest65.sv, 2
   Illegal combination of procedural drivers
   Variable "a" is driven by an invalid combination of procedural drivers.
   Variables written on left-hand of "always_comb" cannot be written to
   by any other process, including other "always_comb" processes.
   This variable is declared at
   slangtest65.sv, 2
   2: logic a;
   The first driver is at
   slangtest65.sv,6
    always_ff @(posedge clk) begin
        if (rst) a <= '0;
    ...
   The second driver is at
   slangtest65.sv, 11
    always_comb begin
        a = 1;
```

MikePopoloski, 2022-02-16:
Thanks for the report. This has been on my list for a long time, and it's probably getting close to being appropriate for me to work on it. It's a pretty complicated problem to solve in a performant manner because you need to track assignments on a per-bit basis, but I have some thoughts about how to go about it.

MikePopoloski, 2022-03-16:
This is done now.

MikePopoloski, 2022-03-16:
Actually, I haven't yet done the rules for always_ff / always_comb blocks so I'll keep this open until that's done too.

MikePopoloski, 2022-03-27:
Should be complete now.

closed 2022-03-27 by MikePopoloski