MikePopoloski/slang #514: Simulation support ??

Blebowski, 2022-02-22:
Hello, I found out in your docs that eventually you would like to add simulation support to slang.

I was wondering what would be the roadmap / directions there? Also, where would you suggest
starting, if I wanted to help with this?

Do you plan implementing System Verilog LRM compliant scheduler ?

I am aware of commercial simulators (Big-3) and GHDL/Verilator, and I know this is insanely difficult
task considering all things proper simulator needs (waveform debug (possibly integration with Gtkwave),
coverage, etc...). I was just wondering whether there is something I could start with...

Btw. cudos to slang. I like the way this open source project is developed. It goes for full language
support, and effective / fast parsing, everything is nicely documented. Lot of "new HDLs/EDA tools" in FOSS
world are trying to re-invent the wheel, connect everything together through some intermediate compiler IIR
layers, which is good from compiler writer perspective, but not really from HW developer perspective. Not to
mention complete lack of real usability (I mean in commercial ASIC/FPGA designs or models) beyond PoF state,
which IMHO is pity. Slang is perfect for this, it is really trying to create usable/complete SW
(even if it is just parser/elaborator for now), not just SW which implements 75 % from 10 different things

From what I have seen, Verilator is by far the best out of OS tools, but it is still not complete since it lacks
proper scheduler, therefore support of SDF annotated simulations is not possible (however, there is ongoing
work towards that). Also, no OS simulator has "proper" expression/statement/branch coverage nor integrated
waveform viewer which would be feature-wise comparable with Simvision or Verdi.

I was wondering whether it would be possible for some OS simulator to reach level of Big-3 simulator
in following features:
 - VHDL / System Verilog / Verilog support
 - VPI / DPI / VHPI support
 - Proper per-instance code coverage (expression, statement, branch, toggle)
 - Good Linting capabilities
 - Good integration with waveform viewer, not just plain VCD dump.
 - Easy to debug simulation (e.g. X tracing, native display of e.g. UVM/OSSVM/OVM objects)

MikePopoloski, 2022-02-23:
Yes, I'd like to have simulation support in slang, with a fully compliant event-based scheduler. As you correctly note, this is a huge undertaking. I have some thoughts about how I'd approach this, and I started writing some of it in slang already, which you can see in the codegen folder. This approach generates bitcode for LLVM, and I got far enough to have the most basic of Hello Worlds run with a $display call and haven't touched it since. If you were going to jump in that's where I'd recommend, but I'd warn that it's not a simple undertaking.

Is it possible for an OSS stack to reach the level of the big three? I think so, but it would require many many developers working together. Think of how GCC and Clang are the best in class development toolchains and they're completely OSS, but they also have many people contributing. On my own I'll never get there, but I'll keep making progress anyway (it's a fun hobby project).

MikePopoloski, 2022-03-01:
I'm going to close this out. Feel free to open a Discussion if you want to discuss further.

closed 2022-03-01 by MikePopoloski