MikePopoloski/slang #516: Functions in Constraints (SV LRM 18.5.12)

jrudess, 2022-03-18:
In LRM 18.5.12, there are a set of restrictions on what kind of functions are allowed in constraints.   Are all of these checks implemented?

- Functions that appear in constraint expressions cannot contain output or ref arguments (const ref is allowed)
- Functions that appear in constraint expressions should be automatic (or preserve no state information) and have no side effects
- Functions that appear in constraints cannot modify the constraints, for example, calling rand_mode or constraint_mode methods.
- Functions shall be called before constraints are solved, and their return values shall be treated as state variables.
- Random variables used as function arguments shall establish an implicit variable ordering or priority.  Constraints that include only variables with higher priority are solved before other, lower priority constraints.  Random variables solved as part of a higher priority set of constraints become state variables to the remaining set of constraints.  (LRM has an example)
- Circular dependencies created by the implicit variable ordering shall reslut in an error.
- Function calls in active constraints are executed an unspecified number of times (at least once) in an unspecified order.

Basically VCS has serious deficiencies properly catching these (it won't catch it in an IP compile, then it will fail to compile in an SoC).  My hope is to put Slang as the front-end for IPs that are consistently having problems violating these rules.

MikePopoloski, 2022-03-18:
slang enforces the first bullet point. It's not clear that the second bullet is a hard error that needs to be diagnosed by the compiler or whether it's just saying you "shouldn't" do that if you want your code to be sane.  Four, five, and seven aren't actually restrictions.

That leaves three and six (and maybe two) that aren't currently implemented; I overlooked them when I did my constraint implementation. It should be architecturally easy to plug them into the existing code, though the actual restrictions themselves may be tricky to implement.

jrudess, 2022-03-18:
Agreed.  I pasted the entire set largely for context, since number 6 refers to the implicit ordering which number 5 describes, but also your understanding of the spec is a lot deeper than mine, so you might notice something I didn't think of.

I believe that #6 is where VCS compile inconsistencies largely come from, with the same code working in one build, and then producing errors in another, or working with one set of elaboration options, then failing with a different set.

jrudess, 2022-03-18:
Depending on the complexity of implementing the restriction, a different option might be to have a warning which can be enabled to flag function calls inside constraints, with the ability to escalate optionally to an error.  We often tell people to put the function calls in pre_randomize() instead as a general practice, so I think some projects would utilize that.

MikePopoloski, 2022-03-18:
I think a check like that would better live in a separate tool like a "slang_tidy" kind of thing (analogue to clang_tidy). We could throw whatever stylistic or project-specific checks we want in there, whereas I think warnings in the compiler proper should be more broadly applicable. It should be relatively easy to leverage the slang library to build such a tool (at least that's the hope).

I will plan on implementing the checks mandated by the standard in the compiler itself, though I may not get to it right away. If you want them faster feel free to take a crack at them yourself -- the jumping off point would be in the ConstraintExprVisitor in Constraints.cpp

MikePopoloski, 2024-05-07:
I came back and looked at this again and realized that you can't reliably enforce this rule at compile/elaboration time; whether or not there are circular dependencies depends on the enabled state of random variables and constraint blocks, which is controllable at runtime with rand_mode and constraint_mode.

It might still be nice to have a warning saying that there *might* be circular dependencies so I'll leave this open but I'm removing the lang-compliance tag since conforming to the LRM can only truly happen at runtime when randomize is called.

MikePopoloski, 2026-03-06:
I added this in 0c3046b6bfd9d0a6d024546e8481c7b91ca2e6ff and be8e7928bdc855a65793899a3c8c184445defcbd