MikePopoloski/slang #520: Support unsized arrays as DPI argument types

MikePopoloski, 2022-04-17:


MikePopoloski, 2022-11-20:
This has been added.

jrudess, 2022-11-21:
Seeing some type mismatches errors for this.

```systemverilog
slangtest97.sv:10:34: error: value of type 'dynamic array of byte' cannot be assigned to type 'unpacked array [] of byte'
        P::write_buf("file.txt", write_buffer);
                                 ^~~~~~~~~~~~
```
```systemverilog
package P;
    import "DPI-C" context function int write_buf(string outfile, byte write_buf[]);
endpackage

class C;
    function main();
        int rc;
        byte write_buffer[];

        P::write_buf("file.txt", write_buffer);
    endfunction
endclass
```

MikePopoloski, 2022-11-22:
Ah, I did not realize dynamic arrays were still allowed here, but upon closer reading of the LRM they're only disallowed in DPI exports, not imports. I will fix.

MikePopoloski, 2022-11-23:
Should be fixed now.

jrudess, 2022-11-23:
Ah, interesting -- I'll need to go through the LRM more.  I didn't include the read_buff function in the first example because I assumed it was just the same issue.  The big three are at least compiling the following code, but I've only tested runtime sim functionality in VCS.

```systemverilog
slangtest98.sv:10:40: error: value of type 'unpacked array [] of byte' cannot be assigned to type 'dynamic array of byte'
        rc = P::read_buf("input.data", read_buffer);
                                       ^~~~~~~~~~~
```
```systemverilog
package P;
    import "DPI-C" context function int read_buf(string outfile, output byte buffer[]);
endpackage

class C;
    function main();
        int rc;
        byte read_buffer[];

        rc = P::read_buf("input.data", read_buffer);
    endfunction
endclass
```

MikePopoloski, 2022-11-23:
Yeah, I overlooked output parameter handling. The conversion goes in the opposite direction.

MikePopoloski, 2022-11-24:
Should be fixed for output args.

closed 2022-11-24 by MikePopoloski