MikePopoloski/slang #533: Add more lint warnings

MikePopoloski, 2022-04-17:
Ideas for things that should be checked:
* Implicit conversions in constexprs that change value
* empty loop/if body
* header guard mismatch
* misleading if else indentation
* upward names
* implicit conversions
* implicit params
* self-assign
* shadowing names
* signedness issues
* switch coverage of enum
* uninitialized vars in constexprs
* unreachable code
* unused variables
* unused typedef
* unused functions
* unused parameters
* unused fields
* posedge of multi-bit value
* latches, blocking assignments in ffs
* dangling else
* string literal conversions
* missing return
* comparisons that always eval to same result
* enums in set membership ops

Sustrak, 2022-07-08:
Hi,
I would like to add the following lint rule: Register only assigned on reset.
The reason is that most synthesis tools will complain about this.

The following code snipped would trigger this warning:
```
logic my_register;
always_ff @(posedge clk or negedge rst) begin
    if (~rst) begin
        my_register <= '0;
    end else begin
        ...
        { other signals but no, my_register }
        ...
    end
end
```

Where it would be the best place to put the code to detect this scenario?

MikePopoloski, 2022-07-09:
I think it would be easiest as a post-processing task once the AST is constructed. You can walk the tree, find all variables and then loop through their drivers by calling getFirstDriver() / getNextDriver() and look at the places it's assigned.

MikePopoloski, 2025-01-04:
From surveying other tools:

From GCC:
-Winfinite-recursion
-Winit-self
-Wmisleading-indentation
-Wsequence-point
-Wshift-count-negative
-Wshift-count-overflow
-Wshift-negative-value
-Wuninitialized
-Wmaybe-uninitialized
-Wbool-operation
-Wduplicated-branches
-Wduplicated-cond
-Wdiv-by-zero
-Wtautological-compare
-Wshadow
-Wtype-limits
-Wdangling-else
-Wlogical-op

From Clang:
-Wabstract-final-class
-Wbitwise-instead-of-logical
-Wbool-operation
-Wcall-to-pure-virtual-from-ctor-dtor
-Wconditional-uninitialized
-Wconstant-logical-operand
-Wdangling-else
-Wdivision-by-zero
-Wfloat-overflow-conversion
-Wfloat-zero-conversion
-Wfor-loop-analysis
-Wheader-guard
-Wincrement-bool
-Winfinite-recursion
-Wlarge-by-value-copy
-Wliteral-range
-Wmisleading-indentation
-Wreturn-mismatch
-Wreturn-type
-Wself-assign
-Wshadow
-Wshift-count-negative
-Wshift-count-overflow
-Wshift-negative-value
-Wshift-overflow
-Wshift-sign-overflow
-Wsometimes-uninitialized
-Wstatic-self-init
-Wstring-conversion
-Wtautological-bitwise-compare
-Wtautological-compare
-Wtautological-constant-compare
-Wtautological-constant-in-range-compare
-Wtautological-constant-out-of-range-compare
-Wtautological-negation-compare
-Wtautological-overlap-compare
-Wtautological-type-limit-compare
-Wtautological-unsigned-zero-compare
-Wtautological-value-range-compare
-Wtype-limits
-Wuninitialized
-Wunreachable-code
-Wunreachable-code-break
-Wunreachable-code-loop-increment
-Wunreachable-code-return
-Wunsequenced

From Verilator:
BLKSEQ - blocking assign in always_ff
CASEOVERLAP
CASTCONST - dynamic cast
CMPCONST
COLONPLUS
EOFNEWLINE
GENUNNAMED
IMPLICIT - implicit net
IMPLICITSTATIC - functions / task lifetime
LATCH
MISINDENT
NOLATCH
NULLPOR
PREPROCZERO
UNSIGNED
VARHIDDEN
WAITCONST

Other:
Unused struct members
Unused modport stuff
Unused clock vars
Unused pattern vars
Unused assertion vars, ports
Unused nettypes
Unused subroutines
Unused class members
Constraint block cycles
Conversions between structs and vectors
Comparing X's with ==, <, etc or types that can't contain X's

closed 2026-10-03 by MikePopoloski