MikePopoloski/slang #544: Clocking block directions more restrictive than other tools

jrudess, 2022-05-06:
With git-bisect I see that adc7bff898fcb5a11344efc83f57e4340a02ef6d introduced some changes to clocking-block driver checks.

A common BFM design pattern I see is that the interface signals have no directions defined, and a clocking block is written that works for a specific side of the interface.   At runtime, a BFM is configured for which side of the link, and that influences which clocking-block it uses to drive the bus.   The original code compiles with the big 3.

```systemverilog
interface I;
    logic clk;

    logic a;
    logic b;

    clocking cb_fabric @(posedge clk);
        output a;
        input  b;
    endclocking

    clocking cb_device @(posedge clk);
        input  a;
        output b;
    endclocking

endinterface

module dut_device(
    input  logic a,
    output logic b
);
endmodule

module top;
    logic a;
    logic b;

    I intf();

    dut_device dut(
        .a (intf.a),
        .b (intf.b)
    );
endmodule
```
```systemverilog
slangtest72.sv:33:13: error: cannot write to 'b' with a continuous assignment since it's associated with an output clocking signal
        .b (intf.b)
            ^~~~~~
slangtest72.sv:14:16: note: referenced here
        output b;
               ^
```

jrudess, 2022-05-06:
Using multiple interfaces was a rats nest leading to significant more code bloat than I originally anticipated.   For those who find this in the future via internet searches, the following alternative looks like a promising (and transparent) solution, but it requires that all clockvars be wires.

```systemverilog
interface I;
    logic clk;

    bit mode;

    wire a;
    wire b;

    logic driver_a;
    logic driver_b;

    assign a = mode  ? driver_a : 1'bz;
    assign b = !mode ? driver_b : 1'bz;

    clocking cb_host @(posedge clk);
        output a;
        input  b;
    endclocking

    clocking cb_device @(posedge clk);
        input  a;
        output b;
    endclocking

endinterface

module dut_device(
    input  logic a,
    output logic b
);
```

MikePopoloski, 2022-05-07:
I'll still plan on adding the compat flag, since VCS allows it.

jrudess, 2022-05-08:
I just upgraded one of the repos hitting this, and VCS 2021.09 now reports a warning for this connections, with a message that it will be upgraded to an error in a future release.   There still might be a good reason to add a backwards compatibility switch, but this guarantees I'll need to fix all my code either way.

MikePopoloski, 2022-05-08:
Ah, in that case I won't bother with a flag. Thanks for the intel.

closed 2022-05-08 by MikePopoloski