MikePopoloski/slang #548: foreach loop segfault after statement refactor

jrudess, 2022-06-14:
For one of my repos, 126140085105f7954fbf6a030ca9384665dce3d9 is the last commit where slang runs successfully.  The next few commits aren't compiling (gcc 11.1, SLES12), but then with commit 15b76d0db6faba9be837cca42db7269a1cdd799c and newer there is a segfault.

The following backtrace is from HEAD of master.  I haven't been able to isolate the segfault to any specific SV code yet while inspecting the backtrace info.   Will update when I can track down the problem-code or find a simple reproducer.

```gdb
0x00000000005cc89e in slang::ForeachLoopStatement::fromSyntax (compilation=..., syntax=..., context=..., stmtCtx=...) at slang/source/binding/Statements.cpp:1619
1619            it->nextIterator = std::exchange(iterCtx.firstIterator, it);
(gdb) bt
#0  0x00000000005cc89e in slang::ForeachLoopStatement::fromSyntax (compilation=..., syntax=..., context=..., stmtCtx=...) at slang/source/binding/Statements.cpp:1619
#1  0x00000000005ca629 in slang::Statement::bind (syntax=..., context=..., stmtCtx=..., inList=<optimized out>, labelHandled=<optimized out>)
    at slang/source/binding/Statements.cpp:172
#2  0x00000000005cb130 in slang::Statement::bindBlock (block=..., syntax=..., context=..., stmtCtx=...) at slang/source/binding/Statements.cpp:320
#3  0x00000000004f9b57 in slang::StatementBlockSymbol::getStatement (this=0x113d2d38, parentContext=..., stmtCtx=...) at slang/source/symbols/BlockSymbols.cpp:44
#4  0x00000000005ca8a3 in slang::Statement::StatementContext::tryGetBlock (target=..., context=..., this=0x7fffffff9ae0) at slang/source/binding/Statements.cpp:70
#5  slang::Statement::bind (syntax=..., context=..., stmtCtx=..., inList=<optimized out>, labelHandled=<optimized out>) at slang/source/binding/Statements.cpp:169
#6  0x00000000005ca9ae in slang::Statement::bindItems (items=..., context=..., stmtCtx=...) at slang/source/binding/Statements.cpp:339
#7  0x000000000053ca7d in slang::SubroutineSymbol::getBody (this=this@entry=0x113d2888) at slang/source/symbols/SubroutineSymbols.cpp:47
#8  0x00000000004db2b9 in slang::DiagnosticVisitor::handleDefault<slang::SubroutineSymbol> (symbol=..., this=0x7fffffffa020)
    at slang/source/compilation/ElabVisitors.h:50
#9  slang::DiagnosticVisitor::handleDefault<slang::SubroutineSymbol> (symbol=..., this=0x7fffffffa020) at slang/source/compilation/ElabVisitors.h:28
#10 slang::DiagnosticVisitor::handle (this=0x7fffffffa020, symbol=...) at slang/source/compilation/ElabVisitors.h:261
#11 0x00000000004db528 in slang::ASTVisitor<slang::DiagnosticVisitor, false, false>::visitDefault<slang::UnpackedStructType> (t=..., this=0x7fffffffa020)
    at slang/source/../include/slang/symbols/ASTVisitor.h:90
#12 slang::DiagnosticVisitor::handleDefault<slang::UnpackedStructType> (this=0x7fffffffa020, symbol=...) at slang/source/compilation/ElabVisitors.h:57
#13 0x00000000004da308 in slang::DiagnosticVisitor::handleDefault<slang::ClassType> (symbol=..., this=0x7fffffffa020) at slang/source/compilation/ElabVisitors.h:29
#14 slang::DiagnosticVisitor::handle (symbol=..., this=0x7fffffffa020) at slang/source/compilation/ElabVisitors.h:134
#15 slang::ASTVisitor<slang::DiagnosticVisitor, false, false>::visit<slang::ClassType> (t=..., this=0x7fffffffa020)
    at slang/source/../include/slang/symbols/ASTVisitor.h:64
#16 slang::Symbol::visit<slang::DiagnosticVisitor&> (this=0x113d2298, visitor=...) at slang/source/../include/slang/symbols/ASTVisitor.h:199
#17 0x00000000004db77e in slang::DiagnosticVisitor::finalize (this=this@entry=0x7fffffffa020) at slang/source/compilation/ElabVisitors.h:334
#18 0x00000000004c4871 in slang::Compilation::getSemanticDiagnostics (this=0x7fffffffaa00) at slang/source/compilation/Compilation.cpp:800
#19 0x00000000004c5677 in slang::Compilation::getAllDiagnostics (this=0x7fffffffaa00) at gcc/11.1.0/include/c++/11.1.0/optional:440
#20 0x00000000004a3cc9 in Compiler::run (this=this@entry=0x7fffffffa7b0) at slang/tools/driver/driver.cpp:363
#21 0x00000000004a6ef1 in driverMain<char**> (argc=<optimized out>, argv=<optimized out>, suppressColorsStdout=<optimized out>, suppressColorsStderr=<optimized out>)
    at slang/tools/driver/driver.cpp:844
#22 0x00007ffff7328765 in __libc_start_main () from /lib64/libc.so.6
#23 0x000000000048c2f9 in _start () at ../sysdeps/x86_64/start.S:118
```

jrudess, 2022-06-14:
Forgot to include the assertion that triggers with debug-build:

internal compiler error: Assertion '!arrayRef.bad()' failed
  in file slang/source/binding/Statements.cpp, line 1594
  function: static slang::Statement& slang::ForeachLoopStatement::fromSyntax(slang::Compilation&, const slang::ForeachLoopStatementSyntax&, const slang::BindContext&, slang::Statement::StatementContext&)

MikePopoloski, 2022-06-16:
Hmm, it's not obvious from the backtrace or assert how we could get into this state. If you're able to break at the assertion in gdb then you should be able to inspect one of the SourceLocations of the syntax nodes involved. In debug mode the SourceLocation objects have a path that will tell you the file and then a byte offset in that file to tell you which code is causing this.

jrudess, 2022-06-17:
The problem code is in the Acellera UVM library.  I haven't setup a simple reproducer yet, but you can replicate with:

```
wget https://www.accellera.org/images/downloads/standards/uvm/UVM-18002-2020-11tar.gz
tar -zxvf UVM-18002-2020-11tar.gz
slang -I1800.2-2020-1.1/src 1800.2-2020-1.1/src/uvm_pkg.sv
```

MikePopoloski, 2022-06-17:
I just noticed that the UVM tests in the sv-tests repo are failing under slang as well, so I assume it's the same problem.

jrudess, 2022-06-17:
All issues resolved.  Thanks!

closed 2022-06-17 by jrudess