MikePopoloski/slang #550: Issue with environment variable expansion for +incdir in filelist

jrudess, 2022-06-20:
I think the _ character is not being treated as a legal character for an enviornment variable name, leading to environment variable expansion being for the wrong name.

```
slang/build/bin/slang -f incdir.f 
warning: include directory '_HOME/src' does not exist
error: no input files
```

And incdir.f contains:
```
+incdir+$UVM_HOME/src
```

And in the shell $UVM_HOME is /home/user/uvm

jrudess, 2022-06-20:
Below patch resolves the test case, but also noticed isValidCIdChar() in Compilation.cpp could be used.

```diff
diff --git a/source/util/CommandLine.cpp b/source/util/CommandLine.cpp
index 22a9e808..e9016b01 100644
--- a/source/util/CommandLine.cpp
+++ b/source/util/CommandLine.cpp
@@ -322,7 +322,7 @@ std::string CommandLine::expandVar(const char*& ptr, const char* end) {
     else if (isAlphaNumeric(c)) {
         std::string varName;
         varName += c;
-        while (ptr != end && isAlphaNumeric(*ptr))
+        while (ptr != end && (isAlphaNumeric(*ptr) || *ptr == '_'))
             varName += *ptr++;
 
         return getEnv(varName);
```

MikePopoloski, 2022-06-20:
Fixed in 864e4139226bd276be0d4d581e0a9b271cd2660f

closed 2022-06-20 by MikePopoloski