MikePopoloski/slang #554: foreach error when iterating through array accessed via class handle from function return value

jcurtiss8086, 2022-06-23:
```
arraytest.sv:15:18: error: name must refer to an iterable array
        foreach (c2.get().a[i]) begin
                 ^~~~~~
arraytest.sv:15:24: error: expected '['
        foreach (c2.get().a[i]) begin
                       ^
arraytest.sv:15:25: error: expected expression
        foreach (c2.get().a[i]) begin
                        ^
arraytest.sv:15:31: error: expected ';'
        foreach (c2.get().a[i]) begin
                              ^
```
```systemverilog
class C1;
    int a[];
endclass

class C2;
    C1 c1;
    function C1 get();
        return c1;
    endfunction
endclass

module top();
    C2 c2;
    initial begin
        foreach (c2.get().a[i]) begin
        end
    end
endmodule
```

MikePopoloski, 2022-06-24:
This isn't legal in SystemVerilog -- according to the LRM foreach doesn't accept an arbitrary expression, only a hierarchical name to an array. Xcelium and Questa both fail to parse this, but it looks like VCS allows it so I could add it as a compatibility extension.

jcurtiss8086, 2022-06-24:
Thanks for feedback - if it's rejected by the other simulators, will consider this a code bug.  It's a pretty minor update to be LRM compliant.

closed 2022-06-24 by jcurtiss8086