MikePopoloski/slang #556: Some crashes from fuzzing

furkanusta, 2022-06-26:
I've did some runs with a fuzzer and there were some problematic cases.

First of all, I must say that none of the cases look like something that can appear during a normal usage, some of them even contain non-ASCII characters. 

INP1: `'module mod(.*);Y`
Output: `internal compiler error: /home/eksi/Projects/Contribute/slang/source/symbols/Scope.cpp:514: Default case should be unreachable!`

INP2: `function tion tion foo(ion foo(o; enfun#pionQf)v)foo(o; enfun#pion#f)v));t`
Output: `Segmentation Fault: slang/source/types/Type.cpp:1074`

Since these two contain non-printable characters, I wasn't able to paste them.
[oom.txt](https://github.com/MikePopoloski/slang/files/8986711/oom.txt)
[timeout.txt](https://github.com/MikePopoloski/slang/files/8986712/timeout.txt)

There are around 10 more inputs found by AFL that is similar to INP2 but they all end up in the same place, that is why I didn't include them.

MikePopoloski, 2022-07-02:
Thanks for the report. I fixed the first two issues. The latter two are less bugs and more of a denial of service by asking the compiler to elaborate huge designs -- I'll try to figure out some mitigations for that.

MikePopoloski, 2022-08-10:
These should all be fixed now.

closed 2022-08-10 by MikePopoloski