MikePopoloski/slang #557: Comments and lines

qarlosalberto, 2022-06-30:
Is it possible show the line number and comments in the ast JSON output?

qarlosalberto, 2022-07-01:
That's cool.

My idea is to use slang in TerosHDL (https://terostechnology.github.io/terosHDLdoc/). Currently I'm using for the parser tree-sitter-verilog (https://github.com/tree-sitter/tree-sitter-verilog)

How you can see I extract the comments in the code to generate the documentation: https://terostechnology.github.io/terosHDLdoc/documenter/start.html

So I would like to do the same using slang instead tree-sitter-verilog

MikePopoloski, 2022-07-01:
Ah nice. You might be better off with using slang as a library and having it dump just the various entities you care about and their comments, since you probably don't care about all the expression / statement trees that will otherwise clog up the JSON. I can provide assistance with that if you want.

qarlosalberto, 2022-07-01:
Perfect! Sadly I'm not familiarized with C++, but I will try it.

MikePopoloski, 2022-08-09:
I added a set of Python bindings along with a (very) simple example of using them to extract comments. Hopefully that helps.

closed 2022-08-09 by MikePopoloski