MikePopoloski/slang #558: Using slang to create a tool

jameshanlon, 2022-07-02:
I'm curious to know what is the anticipated approach to using Slang to create a tool for SV. Either:
- You end up duplicating large parts of the driver.cpp code to create a new driver for a tool.
- Use the existing Slang driver tool to create a representation of the design on disk (ie JSON), but currently there is no way to deserialise an AST.
Or is there a better way?

MikePopoloski, 2022-07-02:
I think you have the current state correctly summarized. If parts of the driver are useful to many possible tools that use slang then perhaps they should be factored out into the library so they can be shared. It's really all command line flags and processing and using those options to configure the various library components. If you have a use case perhaps you can take a look at which parts make sense to factor out?

jameshanlon, 2022-07-03:
Thanks for the quick reply @MikePopoloski. That's good to know, and in which case I'll continue with creating new driver code and keep in mind whether parts can be factored out.

closed 2022-07-03 by jameshanlon

MikePopoloski, 2022-08-09:
FYI I factored out a bunch of the driver tool into a Driver class in the slang library. Hope that helps!

jameshanlon, 2022-08-10:
That's great @MikePopoloski - thanks for letting me know.