MikePopoloski/slang #559: Bit blasting a design

jameshanlon, 2022-07-03:
I have been considering how to go about fully bit blasting a design in order to create a bit-level dataflow graph. Slang seems like an ideal basis from which to do this, but I am wondering what (if any) components may be reusable/useful in handling the evaluation of procedural constructs in always blocks, given that similar work has to be done for generate constructs during elaboration. Any insights would be greatfully appreciated.

MikePopoloski, 2022-07-03:
I'm not entirely sure what you're looking for when you say "bit blasting", but slang can already evaluate most of the things you'd find in a procedural context because it needs to be able to evaluate function calls during constant expression elaboration. You can look at the EvalContext class and the various eval methods on statements and expressions. Of course it doesn't handle anything related to timing or scheduling events, so if you need that you're looking more at implementing a simulator (which is doable but obviously a lot more work).

jameshanlon, 2022-07-03:
I'm not sure if 'bit blasting' is exactly the right term, but what I mean is expanding procedural blocks to remove conditional and looping constructs, into sets of assignments for each atom that is on the left hand side of an assignment. There is an explanation of sorts [here for a synthesis strategy](https://www.cl.cam.ac.uk/teaching/1011/SysOnChip/slides/sp1rtl/zhp9dafa45aa.html), but to give a simple example, the block:
```
always_ff @(posedge i_clk)
 for (i=0; i<8; i++) begin
      if (i == 0)
        data_q[0] <= i_data;
      else 
        data_q[i] <= data_q[i - 1];
 end
```
Would expand to:
```
always_ff @(posedge i_clk) begin
  data[0] <= i_data;
  data[1] <= data[0];
  ...
  data[7] <= data[6];
end
```
What I'm trying to do is be able to determine the bitwise dependencies between variables in a design in order to trace paths. I was anticipating rewriting the AST to convert it into the above form. This is different from a synthesis engine that would lower the AST into some form of IR or netlist. While writing this, it seems clearer that what I'm trying to do is fairly specific and not within the scope of slangs advertised capabilities! But I thought it was worth asking anyway.

MikePopoloski, 2022-07-04:
This seems very accomplishable with slang. It will give you the elaborated AST and you can walk it and unroll the design however you see fit.

jameshanlon, 2022-07-04:
Thanks for the clairification - I shall investigate further. The EvalContext class looks like it will be helpful in doing this.

closed 2022-07-04 by jameshanlon