MikePopoloski/slang #565: Errors with module range specification and implicit port connections

jrudess, 2022-07-20:
The original code is compiling for the big-three.  Have I missed anything in the LRM that disallows this?

```systemverilog
slangtest87.sv:14:10: error: implicit named port 'a' of type 'logic' connects to value of inequivalent type 'logic[7:0]'
        .a,
         ^
slangtest87.sv:15:10: error: implicit named port 'b' of type 'logic' connects to value of inequivalent type 'logic[7:0]'
        .b
         ^
```
```systemverilog
module M(
    logic a,
    logic b
);
endmodule

module top;
    localparam N = 8;

    logic [N-1:0] a;
    logic [N-1:0] b;

    M m [N-1:0] (
        .a,
        .b
    );
endmodule
```

MikePopoloski, 2022-07-20:
The section on implicit port connections doesn't say anything about arrays of instances, just that types need to be equivalent, so I interpreted that as not allowing the instance array slicing mechanism. It seems just as valid to interpret it the other way though, and since the big-three do I can make slang do the same.

MikePopoloski, 2022-07-23:
Fixed in 8ea0131e64692e2894264ea276b1bb7d09a6b704

closed 2022-07-23 by MikePopoloski