MikePopoloski/slang #566: Accessing static member of a forward-typedef class

jrudess, 2022-07-21:
This one compiles in the big-three, but it might be an LRM enhancement.

```systemverilog
slangtest88.sv:7:9: error: cannot resolve scoped class member via incomplete forward typedef
        S::a["test"] = a;
        ^
```
```systemverilog
typedef class S;
typedef class A;

class C;
    function f();
        A a = new();
        S::a["test"] = a;
    endfunction
endclass

class S;
    static A a[string];
endclass

class A;
endclass
```

MikePopoloski, 2022-07-21:
Yeah, this is explicitly disallowed by the LRM in 6.18, with an example and everything.
```
A forward typedef shall be considered incomplete prior to the final type definition. While incomplete
forward types, type parameters, and types defined by an interface based typedef may resolve to class types,
use of the class scope resolution operator (see 8.23) to select a type with such a prefix shall be restricted to a
typedef declaration. It shall be an error if the prefix does not resolve to a class.
```

MikePopoloski, 2022-07-21:
Hmm, although now that I read it again I notice that it only says *types* cannot be selected, nothing about data members. Perhaps that's why the big three support your example.

jrudess, 2022-07-21:
Incidentally earlier versions of VCS explicitly violated the type rule in 6.18, and I've had to fix code that was reference typedefs in forward-typedef classes, usually by moving the typedefs to a higher level package scope.  I believe VCS implemented strict 6.18 type checking sometime around VCS 2019 or VCS 2020.

MikePopoloski, 2022-07-23:
Fixed in e503b2d095ae352151f0112a99302e51444a0ed1

closed 2022-07-23 by MikePopoloski