MikePopoloski/slang #567: DPI and open-array-handles

jrudess, 2022-07-21:
This one compiles in VCS.  Still working on getting Questa/Xcelium clean for this IP to see if it works there.   There's an example of an open-array for an import function H.12.9 and this annex defines svOpenArrayHandle as 'typedef void*'.

```systemverilog
slangtest89.sv:2:70 error: 'dynamic array of int unsigned' is not a valid argument type in a DPI subroutine
    import "DPI-C" context function int unsigned f(input int unsigned data[]);
                                                                      ^
```
```systemverilog
package P;
    import "DPI-C" context function int unsigned f(input int unsigned data[]);
endpackage
```
```c++
unsigned f(const svOpenArrayHandle data) {
}
```

MikePopoloski, 2022-07-21:
Yeah, this is known to not be supported yet; see #520

jrudess, 2022-07-21:
Ah, should have checked open issues first.  Closed in favor of #520.

closed 2022-07-21 by jrudess