MikePopoloski/slang #574: Ignore Encrypted code?

Practical-UVM-Step-By-Step, 2022-08-07:
Would it be possible to have slang/driver have a way that would allow encrypted code to be skipped in lint-only/parsing modes?
was trying to parse some code as a test of the parser, and compilation will fail because the tokens are encrypted. That's what I saw and maybe I missed something in the command line options?

Maybe a warning that it's not complete or something from diagnostics.cpp would be useful too.

The fix to me seemed to be maybe just before Line 484 of SourceManager.cpp by calling a strip_protected() or some other method (I made that one up) or alternately somehow drop it in Compilation/AddScope..

I wasn't sure where would be the best place to put  such a handler.

MikePopoloski, 2022-08-07:
Is this code encrypted via the standard pragma protect system described in the LRM or something bespoke?

Practical-UVM-Step-By-Step, 2022-08-08:
I dont see any other annotations that may suggest if its something weird. .
Essentially all I saw was:

`protected
<some lines of basically several characters
`endprotected

which means that someone encrypted the code before slang saw it.

which according to LRM (Section 34 in  1800-2017, page 918) seems like tools should be able to handle.  Since Slang's tools didnt encrypt it, I'd expect a warning of some sort or whatever,

My ask:
Dont implement it yet. 

Please tell me where you'd put a handler for it first. (maybe you will give some callback function that can decrypt it for you in some other code? )  I may be the first to bring this to your attention, but as the other threads like formatter and others bring interest, this may be something that will come up.

IMO, One place to put it would be the preprocessor to swallow all the tokens between `protected and `endprotected. Dunno if that's the right choice though but it may wind up becoming a pre-processor option.

MikePopoloski, 2022-08-08:
Yeah, the Preprocessor is where this should start. There is a Preprocessor::applyProtectPragma method that right now just issues a warning and does nothing. Probably this feature will also require some integration back into the Lexer, as the encrypted text isn't guaranteed to be valid tokens (or even printable text) so there will need to be a mode to just consume characters, and possibly do decoding based on the input encoding, since you might not be able to reliably detect the end_protectected keyword if there's a binary blob in the middle.

udif, 2022-08-25:
I wrote the support for this stuff for Verilator (https://github.com/verilator/verilator/pull/2886).  

It does not read the encrypted code ofcourse, but it knows how to gracefully skip it, and make sure it is legally constructed (check all encryption header fields, and if the Base64 content length matched the expected declarations). A fully compliant implementation (including the encryption stuff itself) can be not so trivial, as it may contain, if I remember correctly, nested encryption layers. 

As far as I remember, you can legally start encrypting code at any point, but the IP encrypted stuff I saw sometimes leave the port declarations unencrypted and only encrypts the internal module content.
This means that if you ignore the encrypted stuff, you get a nice module blackbox definition. This can be useful.

MikePopoloski, 2022-08-25:
Yeah, that's my plan: do checking of all the pragmas, skip over all of the encrypted code, but otherwise not bother trying to implement the decryption logic. I would if I thought it would ever be useful but there's basically zero chance anyone will ever give out keys that work with open source tools (because it would then be trivial to modify the tool to print out the original source).

udif, 2022-09-05:
I see you have added a few commit for encrypted content. Is that supposed to be ready? I'm still getting warnings on unknown `` `protect`` and `` `endprotect` tags.

MikePopoloski, 2022-09-05:
No, I will close the issue when I think it's finished.

MikePopoloski, 2022-09-07:
I think this should be mostly done now actually. I have one more warning I want to add just to indicate that these regions are being skipped but all of the decoding should be handled now. Please open issues if you run into problems.

MikePopoloski, 2022-09-09:
This should be complete now.

closed 2022-09-09 by MikePopoloski