MikePopoloski/slang #575: Minimum Compiler version in CMakelists...

Practical-UVM-Step-By-Step, 2022-08-07:
Hi: 
Would it be possible to indicate minimum compiler version (CMAKE_CXX_COMPILER_VERSION VERSION_LESS "something" ) 
in your makefiles to whatever you're building with?

(I googled https://stackoverflow.com/questions/14933172/how-can-i-add-a-minimum-compiler-version-requisite)

I was attempting to build on multiple platforms as I'm still learning,  and one thing I noticed is on Ubuntu, G++9 didnt like your code but G++12 seems to hold up ok on my ubuntu box.

Hopefully a good idea to have this check?

Thx

Practical-UVM-Step-By-Step, 2022-08-07:
Just saw your GCC in your buildflow.

closed 2022-08-07 by Practical-UVM-Step-By-Step

MikePopoloski, 2022-08-07:
I support gcc version 8 and above. There's a CI build that verifies things still build with GCC 8, so 9 should have been fine. Can you post what error you were running into?

I'll look into whether it's a good idea to have the CMakeLists also check compiler versions.