MikePopoloski/slang #583: Syntax is being printed incorrectly

bensampson5, 2022-08-12:
I've been playing around with SyntaxTree and SyntaxPrinter and noticed some weird behavior where it doesn't print the port identifier correctly and instead replaces it with "unsigned long long". Not sure if it's a parsing or printing issue. I'm using the latest commit on master:

```
git submodule status
 8cec9dc287fda4cff8a868a874d9588ab611ebb4 ../external/slang (v0.3-1395-g8cec9dc2)
```

I have a main function that does:

```
auto tree = slang::SyntaxTree::fromFile(argv[1]);
std::cout << slang::SyntaxPrinter::printFile(*tree) << std::endl;
```

And given the following input:

```
module test_module (input logic a1, input logic a2, input logic a3, input logic a4, output logic y);
    assign y = a1 & a2 & a3 & a4;
endmodule
```

It replaces the variable "y" with "unsigned long long".

```
module test_module (input logic a1, input logic a2, input logic a3, input logic a4, output logic unsigned long long);
    assign unsigned long long = a1 & a2 & a3 & a4;
endmodule
```

bensampson5, 2022-08-12:
I also tested on the v1.0 tag and it wasn't working on that tag either. Output was the same.

MikePopoloski, 2022-08-12:
That does seem weird. I'm unable to reproduce locally though. "unsigned long long" isn't even a type you'd find in Verilog, so I suspect you've got memory corruption somewhere. Assuming your code isn't doing something you haven't said, are you sure you're linking the slang lib properly to your application? If there was, for example, a mismatch in compiler definitions then it's anybody's guess what the resulting behavior would be.

bensampson5, 2022-08-17:
Ok this is weird. I can no longer recreate it locally either. I guess I may have linked it incorrectly that one time or I also just upgraded my OS which uses a newer version of g++ so not sure if that affected it either. Anyway, it's worked again and I can't be bothered to go back and see if I can recreate it

closed 2022-08-17 by bensampson5