MikePopoloski/slang #593: Segmentation Fault when processing AST in python bindings

meggiman, 2022-08-24:
First of all I want to thank you for this truly awesome work. The parser itself using the CLI really is remarkably robust and reads every RTL I have thrown at it so far. However, the python binding pyslang currently does seem to have a number of issues that prevent me from using it at all. Besides the fact that not all CPP classes are exposed via pybind (I see that you just recently pushed some commits to improve the status quo), I have not been able to traverse the AST in python without triggering segmentation faults. 
Below is a minimal failing case. If I use the library just completely the wrong way I would appreciate a pointer on how to write my own AST Visitors and Transformers in python (I know this should already be possible in CPP though). 

``` python
import pprint
import pyslang
module = """
module sync #(
    parameter int unsigned STAGES = 2,
    parameter bit ResetValue = 1'b0
) (
    input  logic clk_i,
    input  logic rst_ni,
    input  logic serial_i,
    output logic serial_o
);
   (* dont_touch = "true" *)
   (* async_reg = "true" *)
   logic [STAGES-1:0] reg_q;
    always_ff @(posedge clk_i, negedge rst_ni) begin
        if (!rst_ni) begin
            reg_q <= {STAGES{ResetValue}};
        end else begin
            reg_q <= {reg_q[STAGES-2:0], serial_i};
        end
    end
    assign serial_o = reg_q[STAGES-1];
endmodule
"""
st = pyslang.SyntaxTree.fromText(module)
def ast_to_dict(ast_node):
    match ast_node:
        case pyslang.SyntaxNode() as ast_node:
            return {
                'kind': ast_node.kind.name,
                'children': [ast_to_dict(sub_node) for sub_node in ast_node]
            }
        case pyslang.Token() as token:
            return token.valueText
ast_dict = ast_to_dict(st.root)
pprint.pprint(ast_dict)
```  

Expected Behaviour: A pretty printed AST in python dictionary form.

Actual Behaviour: Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)

MikePopoloski, 2022-08-25:
Thanks for the issue report. There's nothing particularly weird about your code, so I'll have to reproduce on my end and try to debug it. pybind11's lifetime semantics are very tricky so it's very likely this is just a bug in the binding logic somewhere.

MikePopoloski, 2022-08-26:
This should be fixed by 987b026bc536fc4c731c5aa7b8f3d8d68a743eec

closed 2022-08-26 by MikePopoloski