MikePopoloski/slang #597: Question: How to use slang AST for file dependency tree extraction

meggiman, 2022-08-26:
I am trying to develop some helper script on the basis of pyslang that extracts all file inter-dependencies from a given SystemVerilog project. I am concentrating mainly on the synthesizable subset of the language. So in essence, given an elaborated top-level module I want to derive the tree of:
- file inclusions
- explicit package imports
- implicit package imports (i.e. usage of constants/types through their fully-qualified-name)
- module/interface Instantiations

 However, I don't see how to efficiently do that with the given data structure/convenience functions exposed by slang/pyslang. 

I.e.:
- Is there any means to get the full file inclusion tree from the preprocessor? I could not find any field in the Preprocessor or SourceManager class that could help me directly. A naive approach would be to traverse the entire AST and query the SourceManager from which file a particular token originates. But then again, the individual tokens do not store their SourceLocations. Only the SyntaxNodes store their range and that granularity might not be sufficient (i.e. if a single value token originates from a preprocessor macro).
- Is there a simple way to traverse the elaborated *symbol tree* in python? The best I could find currently was to write child traversal logic manually for every subclass of `Symbol` since the `Symbol` base class itself is not iterable. For `SyntaxNode` this is relatively simple since the base class provides the iterator. This would be very useful to extract e.g. all the Type-, Value- and `InstanceBaseSymbols` from the `RootSymbol` without having to care about how to handle all the possible intermediate nodes.

MikePopoloski, 2022-08-26:
1. There isn't an easy way right now but it would not be hard for me to add it -- the relevant classes (Preprocessor and SourceManager) already know this information, so it's just a matter of exposing it.

2. The `Symbol` class itself is not iterable but `Scope` is, and all symbols that can have children inherit from `Scope` so it should work just fine to write something similar to what you have already done for the concrete syntax tree.

MikePopoloski, 2022-08-26:
Also wanted to correct something:

> But then again, the individual tokens do not store their SourceLocations. Only the SyntaxNodes store their range and that granularity might not be sufficient (i.e. if a single value token originates from a preprocessor macro).

This is not correct; Tokens store locations, accessible via the `location` property in pyslang. SyntaxNodes actually do not store their locations -- when you ask for the source range the node will traverse down to the first and last leaf Token and use their locations to form the range.

meggiman, 2022-08-26:
@MikePopoloski Thanks a lot for your quick answers. The hint about leveraging scope is quite helpful. And sorry for spreading false information regarding Tokens. I got a little bit misled by the API documentation. It seems like all fields without Doxygen comments (even though they are public) do not appear in the API documentation on sv-lang.com. Thanks for the correction.

MikePopoloski, 2022-08-26:
Yeah, the doxygen reference needs some love, for sure.

MikePopoloski, 2022-08-30:
I added a SourceManager::getAllBuffers() method. Using this you can loop over all BufferIDs and try querying them with SourceManager::getIncludedFrom(), which should let you build up the chain of includes. Let me know if that's not sufficient for some reason.

closed 2022-08-30 by MikePopoloski