MikePopoloski/slang #598: Unittest fail on  DriverTests for windows environment

HungMingWu, 2022-08-28:
Here is the failed case.

``` cpp
TEST_CASE("Driver file preprocess") {
    auto guard = OS::captureOutput();

    Driver driver;
    driver.addStandardArgs();

    auto filePath = findTestDir() + "test.sv";
    const char* argv[] = { "testfoo", filePath.c_str() };
    CHECK(driver.parseCommandLine(2, argv));
    CHECK(driver.processOptions());
    CHECK(driver.runPreprocessor(true, false));

    auto output = OS::capturedStdout;
    CHECK(startsWith(output, R"(
module m;
    // hello
    string s = ")"));

    CHECK(endsWith(output, R"(";
    begin end
endmodule

)"));
}
```
Linux works properly, I think it is related to CRLF issue.
,

MikePopoloski, 2022-08-28:
It works fine locally for me on Windows and in the CI build. Perhaps you have your git configured to mess with line endings when you check out?

Regardless, I can change the test so it's not so sensitive to the line endings.

MikePopoloski, 2022-08-31:
Fixed in 97b2b5e1f2ac83cca1c47095e773c24fc3cd1fe2

closed 2022-08-31 by MikePopoloski