MikePopoloski/slang #600: lexerStack() management bug?

udif, 2022-08-30:
I've been debugging why I'm getting `exceeded max include depth` on my large design, and I think I found the reason:
In `Driver.cpp`:
```
bool Driver::runPreprocessor(bool includeComments, bool includeDirectives) {
    BumpAllocator alloc;
    Diagnostics diagnostics;
    Preprocessor preprocessor(sourceManager, alloc, diagnostics, createOptionBag());

    for (auto it = buffers.rbegin(); it != buffers.rend(); it++)
        preprocessor.pushSource(*it);
```

But `pushSource` is doing this:
```
void Preprocessor::pushSource(SourceBuffer buffer) {
    ASSERT(buffer.id);

    std::cout << "pushSource:  lexerStack.size(): " << lexerStack.size() << std::endl;
    lexerStack.emplace_back(std::make_unique<Lexer>(buffer, alloc, diagnostics, lexerOptions));
}
```

It is creating a lexer stack entry, even when the buffer comes from a list of buffers on the command line (or `-f` which is effectively the same), so my design begins with a lexerStack.size() of more than 3600.

If this is intentional, and lexerStack is not only for nesting `` `include`` files, then why is this check?
```
        else if (lexerStack.size() >= options.maxIncludeDepth)
            addDiag(diag::ExceededMaxIncludeDepth, fileName.range());
```

I see that `options.maxIncludeDepth` is set to 1024, which is what kills my design.

MikePopoloski, 2022-08-30:
Ah yes, good debugging. The lexer stack is also used to implement the "single compilation unit" functionality by pushing all of the sources onto the stack, but of course those should not count toward the include file limit because they aren't include files (and aren't actually nested so there's nothing to protect against). I can fix that.

udif, 2022-08-30:
While we're at it, I was looking at the same area , and saw this:
```
        else if (includeOnceHeaders.find(buffer.data.data()) == includeOnceHeaders.end()) {
            pushSource(buffer);
        }
```
From the little I guess (Haven't done a serious search for this in the code yet), you probably implement something like `` `pragma once`` .
1. I was not aware of such a thing in other Verilog tools
2. What I learn from Wikipedia regarding `#pragma once` for C/C++, they typically match the filename while you match against actual file data?!

MikePopoloski, 2022-08-30:
Yes, slang has a ``` `pragma once``` directive. The SourceManager takes care of deduplicating requests to load files into memory, so checking here for the buffer pointer should be sufficient for determining whether we've already processed a given include.

udif, 2022-08-30:
I'm just pointing at the difference between what `#pragma once` does vs. your implementation (same file in different location would be reloaded by `#pragma once` but not by your code, OTOH, you always read all files even if you skip them, while `#pragma once` reads the 1st instance, but skips the rest at the filename level without even reading them).
(In practice I don't think these differences has any implication in terms of correctness).

I also noticed you have some locking code there. Are you loading files in parallel?

MikePopoloski, 2022-08-30:
SourceManager dedups based on filename, not contents, so it should still behave as you say. There could be differences if the same file is symlinked into different locations in the source tree with different names, but that seems like a distinction not worth getting into.

slang doesn't currently load in parallel but that's the plan -- parsing should be fully independent save for the SourceManager interaction, so that's why there's locking in that one place only. Of course that can't ever apply if you use --single-unit, since you don't know how to parse later files without fully parsing earlier ones, but for projects that make files independent from each other it could be a nice speedup.

udif, 2022-08-30:
> SourceManager dedups based on filename, not contents, so it should still behave as you say.

From what I see here:
```
        else if (includeOnceHeaders.find(buffer.data.data()) == includeOnceHeaders.end())
            pushSource(buffer);
```
It looks like dedups based on actual file data.
When I added debug prints of `buffer.data.data()`  when the condition above was true, I saw actual file data in my logs, not filenames.

> There could be differences if the same file is symlinked into different locations in the source tree with different names, but that seems like a distinction not worth getting into.

I can think of a scenario where this would produce different behavior:

`define1.v` containing:
```
`define XXX 0
```
and symlinked to `define2.v`

and finally, `module.v` that uses `` `XXX``
Then:
```
slang define1.v +define+XXX=1 define2.v module.v`
```
Would behave differently whether you compare file content or filenames.

MikePopoloski, 2022-08-30:
The deduplication happens inside SourceManager, when readHeader() is called. It will normalize the path and avoid rereading the file from disk if it's already been read previously. The Preprocessor isn't comparing the contents of the file data, it's comparing the *pointer values* directly since it's literally the same address in memory.

MikePopoloski, 2022-08-30:
And I'm not sure I follow your example, since there's neither any include directives nor a pragma once, but for sure it's possible to defeat the pragma once detection by symlinking since pragma once only works on file names, so if you change the names they will be considered different.

udif, 2022-08-30:
I'm working on an actual example that will actually work (the one above has several issues, such as module repeated twice but not by `` `include`, and the fact that slang doesn't like macro redefinitions).

udif, 2022-08-30:
> but for sure it's possible to defeat the pragma once detection by symlinking since pragma once only works on file names, so if you change the names they will be considered different.

If it will read it twice for different filenames which are symlinked, then this is the standard behavior `#pragma once` .

I think I got it:
```
../include/slang/parsing/Preprocessor.h:    flat_hash_set<const char*> includeOnceHeaders;
```
When I printed `buffer.data.data()` to `std::cout`, it dereferenced a `const char *` returned by `buffer.data.data()` so it showed the data.
When I call `includeOnceHeaders.find(buffer.data.data())` , it is searching for that **pointer** among the `includeOnceHeaders` set, **not** the c_cstr pointed by it.
If these were two different filenames, they would not match in sourceManager, and would be allocated different text buffers even if their content is identical, and therefore wouldn't match on the `find` call above.

MikePopoloski, 2022-08-30:
Yes, that's right.

udif, 2022-08-30:
After temporarily disabling the `` `include`` depth check , my codebase fails on two issues:
1. redefined modules (solved originally by a vendor's `-allowredefinition` flag)
2.  `` `protect`` , `` `endprotect``

I can start taking a look at these (provided you will accept a PR for the 1st).
I also started taking a look at (2) earlier today, but I'm not too familiar with your parser/lexer tables in `syntax.txt` (not sure how to define those directives that have no additional token - couldn't find an example for that in the other directives)

MikePopoloski, 2022-08-30:
I can take a look at implementing the protect directives. I'll accept a patch for #1, but not sure what the desired behavior you intend? How do you know which module to use for any given instantiation if they share the same name?

udif, 2022-08-30:
I need to check that, but given the vendor's flag name, I assume that a new module definition simply overrides the previous definition (I can easily write a small testcase to verify that).
Given that all tools parse first and elaborate later, I would assume the expected behavior is that all module instances would be from the last definition, that overrode the original one, even for instances inside modules that were defined earlier.

Such a behavior would ofcourse only be triggered when a new flag is specifically enabled (`--allowredefinition` ??)

MikePopoloski, 2022-08-30:
Probably `--allow-redefinition` to match other commands slang already has.

udif, 2022-08-30:
I ran a quick test using 3 files:
```
// a.v
module a;
b b_inst (.x(1'b0));
endmodule

// b0.v
module b;
endmodule

// b1.v
module b(input x);
endmodule
```

When running `vendor-tool b1.v a.v` everything compiles.
When running `vendor-tool b0.v a.v` compilation fails due to missing port in `b`.
When running `vendor-tool b0.v a.v b1.v` compilation fails due to redefinition of module `b`.
When running `vendor-tool -allowredefinition b0.v a.v b1.v` everything compiles again.

MikePopoloski, 2022-08-31:
The include stack issue should be fixed in 5264419368660408ce8e28a4412a5c51aba9cce8

closed 2022-08-31 by MikePopoloski