MikePopoloski/slang #607: Compiling Issues

joeya20, 2022-09-07:
Hi,

This is a bit of a dumb question but I am not sure what to include when compiling a simple test I made. What should directives should I use when compiling? Is there a library that I need to include with -L (I have tried looking with no luck)?

Thanks!

MikePopoloski, 2022-09-07:
If you're using cmake you can do something like `target_link_libraries(driver PRIVATE slangcompiler)`

If you're doing it manually then you probably want to add `-lslangcore -lslangcompiler -lslangparser` as linker inputs.

It's on my TODO list to write some documentation about integrating the library but haven't gotten to it yet.

joeya20, 2022-09-08:
Thanks for the quick response. I have tried that but still no luck.

Here's a screenshot of what I tried and the error I get.
[image]

And here is the test I'm trying:
[image]

MikePopoloski, 2022-09-08:
I think this is due to the ordering of options you've provided to g++. Your test.cpp needs to come first in the list so that the symbols the linker needs to import from later libraries is known. https://eli.thegreenplace.net/2013/07/09/library-order-in-static-linking is a good overview of the topic.

joeya20, 2022-09-11:
wow... thanks!

closed 2022-09-11 by joeya20