MikePopoloski/slang #608: Malloc assertion failure when using as library in cmake

Jomit626, 2022-09-08:
I am trying to using slang as library in my project but an weird malloc assertion occurs.
first, i clone slang into `thirdparty/slang` and use cmake to build my project:
``` cmakelist
cmake_minimum_required(VERSION 3.10.0)
project(slang-link VERSION 0.0.0)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
set(CMAKE_CXX_STANDARD 17)
# svlang
add_subdirectory(thirdparty/slang)
# slang-link
aux_source_directory(src SOURCES)
add_executable(slang-link ${SOURCES})
target_link_libraries(slang-link PUBLIC slangcompiler)
```
then, i have a `main.cpp` in `src` folder:
``` cpp
#include <iostream>
#include "slang/syntax/SyntaxTree.h"
#include "slang/compilation/Compilation.h"
int main(int argc, char *argv[]) {
  auto syntax_tree = slang::SyntaxTree::fromText("\
module test_module (input wire clk, output wire Q, input wire D);\
endmodule\
");
  auto compilation = std::make_unique<slang::Compilation>();
  compilation->addSyntaxTree(syntax_tree);
  auto &diags = compilation->getAllDiagnostics();
  if (diags.empty()) {
    printf("no diags\n");
  }
  return 0;
}
```
here is how i build and run it:
``` bash
mkdir build && cd build
cmake .. -DCMAKE_BUILD_TYPE=Debug && make
./slang-link
```
there is no error during building, but there is a malloc assertion failure when running
``` bash
*[main][~/src/slang-link-test/build]$ ./slang-link
Fatal glibc error: malloc assertion failure in _int_malloc: (unsigned long) (size) >= (unsigned long) (nb)
[1]    6832 IOT instruction (core dumped)  ./slang-link
```
backtrace in gdb:
``` text
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff7b0b543 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff7abb998 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff7aa553d in __GI_abort () at abort.c:79
#4  0x00007ffff7aff67e in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7c20ae0 "Fatal glibc error: malloc assertion failure in %s: %s\n")
    at ../sysdeps/posix/libc_fatal.c:155
#5  0x00007ffff7b18cb8 in __malloc_assert (function=0x7ffff7c21b10 <__PRETTY_FUNCTION__.7> "_int_malloc", line=4298, file=<synthetic pointer>,
    assertion=0x7ffff7c21680 "(unsigned long) (size) >= (unsigned long) (nb)") at malloc.c:299
#6  _int_malloc (av=av@entry=0x7ffff7c5bba0 <main_arena>, bytes=bytes@entry=184) at malloc.c:4298
#7  0x00007ffff7b195c2 in __GI___libc_malloc (bytes=bytes@entry=184) at malloc.c:3315
#8  0x00007ffff7e1847d in operator new (sz=184) at /usr/src/debug/gcc/libstdc++-v3/libsupc++/new_op.cc:50
#9  0x0000555555591008 in std::make_unique<slang::NetType, slang::NetType::NetKind, std::basic_string_view<char, std::char_traits<char> >, slang::Type&> ()
    at /usr/include/c++/12.1.1/bits/unique_ptr.h:1065
#10 0x0000555555573bfe in slang::Compilation::Compilation (this=0x555555d89650, options=...)
    at /home/jomit/src/slang-link-test/thirdparty/slang/source/compilation/Compilation.cpp:104
#11 0x00005555555724e1 in std::make_unique<slang::Compilation> () at /usr/include/c++/12.1.1/bits/unique_ptr.h:1065
#12 0x0000555555571f4f in main (argc=1, argv=0x7fffffffde88) at /home/jomit/src/slang-link-test/src/main.cpp:9
```
valgrind report:
``` text
*[main][~/src/slang-link-test/build]$ valgrind ./slang-link
==12232== Memcheck, a memory error detector
==12232== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
[... 97 lines cut ...]
==12232==    by 0x144821: ska::flat_hash_map<slang::Scope const*, slang::Symbol const*, slang::Hasher<slang::Scope const*>, std::equal_to<slang::Scope const*>, std::allocator<std::pair<slang::Scope const*, slang::Symbol const*> > >::flat_hash_map() (flat_hash_map.hpp:1331)
==12232==    by 0x1273D0: slang::Compilation::Compilation(slang::Bag const&) (Compilation.cpp:44)
==12232==    by 0x1264E0: std::__detail::_MakeUniq<slang::Compilation>::__single_object std::make_unique<slang::Compilation>() (unique_ptr.h:1065)
==12232==    by 0x125F4E: main (main.cpp:9)
client stack range: [0x1FFEFF0000 0x1FFF000FFF] client SP: 0x1FFEFFFB60
valgrind stack range: [0x100402E000 0x100412DFFF] top usage: 13056 of 1048576

Note: see also the FAQ in the source distribution.
It contains workarounds to several common problems.
In particular, if Valgrind aborted or crashed after
identifying problems in your program, there's a good chance
that fixing those problems will prevent Valgrind aborting or
crashing, especially if it happened in m_mallocfree.c.

If that doesn't help, please report this bug to: www.valgrind.org

In the bug report, send all the above text, the valgrind
version, and what OS and version you are using.  Thanks.
```
building env:
Manjaro Linux 21.3.7
gcc 12.1.1
cmake 3.23.3

i have built and ran the `bin/unittest` of slang, and it passes without errors.

MikePopoloski, 2022-09-08:
Your example looks pretty straightforward and how I would expect it to work. I assume the failure is due to some ABI mismatch because of differing compile flags. I'll try to reproduce locally and fix the cmake script to make this just work out of the box.

It's on my TODO list to have a test project using add_subdirectory to ensure that this usage always works.

MikePopoloski, 2022-09-09:
OK yeah, you need `add_compile_options(-DDEBUG)` when building a debug build in your CMakeLists right now. I'll work on making it so you don't need to do that explicitly.

Jomit626, 2022-09-09:
> add_compile_options(-DDEBUG) when building a debug build in your CMakeLists
thank you, this fixes everything.

btw, I want to mention that the binary output is huge (up to 60MB) even when building a release build. I can add ``

MikePopoloski, 2022-09-09:
If you're using the install targets provided by cmake, you can build the `install/strip` target. Otherwise just use the platform's `strip` tool yourself on the output. If you want that at build time always you could add_custom_command in your cmake or something.

MikePopoloski, 2022-09-13:
This should be resolved now.

closed 2022-09-13 by MikePopoloski