MikePopoloski/slang #610: Width-mismatches on port-connections have no warning

jrudess, 2022-09-12:
A common coding mistake is typos in port-list connections which create implicit 1-bit signals.  Made this mistake, and noticed that slang didn't catch the issue, but other simulators flagged warnings.

Would you expect that any of the existing width-mismatch checks should catch the data1/data2 mismatches in this example?  I didn't see a matching issue, or mention of this case in #533.

```Systemverilog
module m(
    input logic       clk,
    input logic [7:0] data1,
    input logic [7:0] data2
);
endmodule

module top;
    logic clk;
    logic [2:0] data2;

    m m(
        .clk   (clk),
        .data1 (data),
        .data2 (data2)
    );
endmodule
```

MikePopoloski, 2022-09-12:
There is no warning for the implicit nets (though I could add one pretty easily), but Wwidth-expand will catch this:
```
test.sv:14:17: warning: implicit conversion expands from 1 to 8 bits [-Wwidth-expand]
        .data1 (data),
                ^~~~
test.sv:15:17: warning: implicit conversion expands from 3 to 8 bits [-Wwidth-expand]
        .data2 (data2)
                ^~~~~
```

jrudess, 2022-09-12:
I missed that -Width-expand wasn't included in -Wextra.  Ran a few repos with it enabled, and probably wouldn't want it in -Wextra.

It might be useful if -Width-expand could be separated for assignments/function-args vs port-connections. 
 For module/interface port-connections, the width mismatch is more-often than not a real bug, while it usually is not for the other cases (converting smaller logic vars to int vars).

MikePopoloski, 2022-09-12:
Yeah, that seems reasonable. Also maybe worth having a dedicated warning for implicit nets that connect to something that isn't 1 bit wide, since that seems pretty dubious in almost all cases?

jrudess, 2022-09-12:
Agreed -- even if a separated width-expand warning is enabled just for ports, it still might not make it obvious that there is a typo in the port-connection name, so a dedicated implicit net warning would help provide better debug info.

MikePopoloski, 2023-03-25:
Added new warnings in 12f466d944bc96d050567921326694cccf98f05e

closed 2023-03-25 by MikePopoloski