MikePopoloski/slang #613: redefining UDPs

udif, 2022-09-13:
If I'll prepare a new PR, similar to #601, but for UDPs , does it have any potential to be accepted?
(Same flag, same warning note as the PR).

MikePopoloski, 2022-09-13:
Yes, should be straightforward to include UDPs.

udif, 2022-09-14:
I've prepared some changes to allow UDP redefinition:
```
@@ -508,15 +510,19 @@ const PrimitiveSymbol& Compilation::createPrimitive(const Scope& scope,
                                                     const UdpDeclarationSyntax& syntax) {
     auto& prim = PrimitiveSymbol::fromSyntax(scope, syntax);
     if (!prim.name.empty()) {
-        auto [it, inserted] = udpMap.emplace(prim.name, &prim);
-        if (!inserted) {
-            auto& diag = root->addDiag(diag::Redefinition, prim.location);
-            diag << prim.name;
-            diag.addNote(diag::NotePreviousDefinition, it->second->location);
-        }
+        if (auto ps = getPrimitive(prim.name))
+            reportRedefinition(*root, prim, *ps, diag::DuplicateDefinition);
         else if (auto defIt = topDefinitions.find(prim.name); defIt != topDefinitions.end()) {
             reportRedefinition(*root, prim, *defIt->second.first);
         }
+        udpMap[prim.name] = &prim;
     }

     return prim;
```
However, when dumping the JSON, I see both the old and new UDP as part of the `members` array, which did not happen when I did the same with modules. I guess I need to look for calls to `Scope::insertMember` , but I haven't dug into this yet.
Also, I see no function to remove a member, I guess slang is only designed to build a tree, not to modify an existing tree, correct?

MikePopoloski, 2022-09-15:
The AST is designed to be functionally immutable, so no changes after construction. I haven't had to support a use case of modifying the tree just yet, but I would envision it as deriving a new tree on top of the existing one. The C# Roslyn compiler is a good example of immutable trees with edits applied on top in a performant way, and it's what I've based some of my design on.

The Primitive symbols will be added to the scope's member list but they aren't resolvable by name since they're in their own namespace according to the LRM (implemented by the canLookupByName method in the source).

udif, 2022-09-18:
OK, created PR #618 as above.
I still don't understand the fundamental implementation difference between duplicate modules and duplicate primitives, e.g. why a similar change for modules produces a JSON where only the last definition remains, while doing the same with a primitive yields a JSON with both copies.

MikePopoloski, 2022-09-18:
Module / interface / program definitions are not themselves symbols, because of the possible presence of parameters there is special handling to create a symbol out of them whenever an instance is needed. Primitives don't have parameters so there's no reason not to make them a symbol like everything else, so they end up in the Scope member list as usual.

Anyway, I merged your PR so I think this issue can be closed.

closed 2022-09-18 by MikePopoloski