MikePopoloski/slang #615: Unrecognized preprocessor directives

udif, 2022-09-14:
IEEE1800-2017 (Appendix E) has a few (optional) preprocessor directives that are as far as I can tell are unrecognized:
```
`delay_mode_distributed
`delay_mode_unit
`delay_mode_path
`delay_mode_zero
`default_trireg_strength
`default_decay_time
```
In addition, I ran into the following, which apparently originate in Verifault-XL:
```
`enable_portfaults
`disable_portfaults
`suppress_faults
`nosuppress_faults
```

Generally speaking, almost all of these are noncritical because they can be ignored by defining these as empty macros.
The only ones which might be problematic are `` `default_trireg_strength `` and `` `default_decay_time `` because they have an extra parameter (non parenthesized), so it can't be eliminated by a macro definition. Fortunately for me, these are not used in my codebase.

udif, 2022-09-14:
There are a few more [here](http://www.pldworld.com/_hdl/2/_ref/se_html/manual_html/c_vlog29.html), but I don't have any documentation for them.
Generally speaking, any nonstandard compiler directive without parameters can be eliminated by a macro definition and is not worth supporting.

Another option might be to add an `--ignore-preprocessor-keyword <preprocessor_keyword_name>` command line flag that will build a list of preprocessor keywords to be ignored along with any extra parameters (until the end of the line), and let users build their own ignore list.

MikePopoloski, 2022-09-15:
I'm ok with adding a bunch of ignores for the nonstandard directives, basically the same as ModelSim has done according to those docs you linked. It's not particularly hard to do.

udif, 2022-09-19:
Created #620 . No tests yet, I assume there will be some iterations.

udif, 2022-10-12:
This issue can be closed, as #620 was merged.

closed 2022-10-13 by MikePopoloski