MikePopoloski/slang #616: No error checking on UDP signal types

udif, 2022-09-14:
(Not a very important bug, but I ran into it when checking some testcases in edaplayground so filed it anyway.)

```
primitive p (out2, in2);
    output out2;
    input in2;

    table
       //   in2     out2
            1   :   0;
            0   :   1;
            x   :   x;
            z   :   x;
            j   :   k;
    endtable
endprimitive

module top;
  wire o;
  p p_inst(o, 1'b0);
endmodule
```
Is accepted although `z` is actually illegal (IEEE 1800-2017 29.3.5).
In fact, any input or output value is accepted (see last line in table).

MikePopoloski, 2022-09-15:
Yes, rules around primitives are not fully implemented yet, which is why support for them is only listed as "partial" on the language support page.

HungMingWu, 2023-01-31:
Have any plan to implement UDP body?

MikePopoloski, 2023-03-25:
Closing as duplicate of #541

closed 2023-03-25 by MikePopoloski