MikePopoloski/slang #622: Adding support for $inferred_clock and $inferred_disable.

automatic-robot, 2022-09-22:
**Describe the bug**
The above two system calls arent recognized by Slang. 

Note: the below snippet is from the LRM on page 468/469  Section 16.14.7

**To Reproduce**
module m(logic a, b, c, d, rst1, clk1, clk2); logic rst;
default clocking @(negedge clk1); endclocking default disable iff rst1;
property p_triggers(start_event, end_event, form, clk = $inferred_clock, rst = $inferred_disable);
@clk disable iff (rst)
(start_event ##0 end_event[->1]) |=> form;
endproperty
property p_multiclock(clkw, clkx = $inferred_clock, clky, w, x, y, z); @clkw w ##1 @clkx x |=> @clky y ##1 z;
endproperty
a1: assert property (p_triggers(a, b, c));
a2: assert property (p_triggers(a, b, c, posedge clk1, 1'b0) );
endmodule

**Additional context**
More interestingly, this doesnt show up in the Section 20 of the LRM or elsewhere which may be why it got missed.
Could you add it?

MikePopoloski, 2022-09-22:
Yes, support for this has not been added yet. This is called out on the language support page for section 16.14.7: https://sv-lang.com/language-support.html#autotoc_md17

This is also a dup of #523 

The inferred value system functions have unique and special logic surrounding them, so it's not as easy as just adding the definitions for them.

MikePopoloski, 2022-11-23:
These have been added.

closed 2022-11-23 by MikePopoloski