MikePopoloski/slang #623: Build configuration with `cmake>=3.24` fails

paulsc96, 2022-09-26:
After installing a CMake version newer than 3.24 (in my case `3.24.2-1` on Arch), build configuration on master (2eac3711f5ef07b6f238194f635d75c70219a4ee) fails with the following error:

```
CMake Error at external/CMakeLists.txt:51 (install):
  install TARGETS given target "fmt" which does not exist
```

Downgrading Cmake to 3.23 (In my case `3.23.3-1`) resolves the issue without making any other changes to the system.

I presume the issue is related to this conditional here:

https://github.com/MikePopoloski/slang/blob/2eac3711f5ef07b6f238194f635d75c70219a4ee/external/CMakeLists.txt#L7

But I am not familiar enough with the build setup to understand what causes the issue.

MikePopoloski, 2022-09-27:
Thanks for the report. This is trying to allow users to take advantage of FetchContent's new feature (as of 3.24) of searching for system-installed packages first before downloading remote content. It's possible I did it wrong though, so I'll try to reproduce locally. In the meantime if you've installed one of these packages into a local cmake normally searches perhaps it's finding that and it's not providing a full cmake target for some reason?

wendajiang, 2022-09-30:
Same Issue. Respect the progress.

wendajiang, 2022-09-30:
After research, In my environment, the problem is that as use the `FetchContent_Declare` and `FIND_PACKAGE_ARGS`, so the cmake system first use `find_package` and `fmt_FOUND` is set to true so that not fetch the target `fmt` from github. However, singlely use `find_package` in my environment, although the variable `fmt_FOUND` is true, like `fmt_LIBRARY` is empty. So `install` command use the target `fmt` is incorrect.

BTW.
[image]
The fmt-config.cmake export two library `fmt::fmt` `fmt::fmt-header-only`,  maybe no `fmt` target ? But I'm not sure about this.

FYI @MikePopoloski

MikePopoloski, 2022-09-30:
Hmm, that does seem like a good clue, thanks for looking.

MikePopoloski, 2022-10-02:
I was able to reproduce by upgrading to Ubuntu Kinetic, which has libfmt 9 installed system wide. I put in a fix so hopefully things work for you now.

closed 2022-10-02 by MikePopoloski