MikePopoloski/slang #632: slang warning changes between versions

udif, 2022-10-10:
**Describe the bug**
I have a strange difference in slang output between 8809f2cc51dedcda350a22559c568344e6438695 and f148cdd4869e4cff368546fbd8365ebfc1b18c3c . I have 4 warnings on a vendor's library file of this form:
(identifier names changed to generic names)

```
some_long_vendor_library_file.sv:2394:15: warning: output net port 'some_port_name' coerced to 'inout' [-Wport-coercion]
  .second_port_name(some_port_name),
                ^~~~~~~~~~~~~~~~
some_long_vendor_library_file.sv:some_line_number:20: note: driven here
 .third_port_name(some_port_name),
                    ^~~~~~~~~~~~~~~~
```
The difference between the two compiled versions is that `third_port_name` and  `some_line_number` are different. both line exists in the file, but different lines are selected each time.

Another slightly different variant is:
```
some_long_vendor_library_file.sv:2658:15: warning: output net port 'some_port_name' coerced to 'inout' [-Wport-coercion]
  .second_port_name(some_port_name),
                ^~~~~~~~~~~~~~~~
some_long_vendor_library_file.sv:some_line_number:20: note: driven here
 .third_port_name(some_port_name[index1]),
                    ^~~~~~~~~~~~~~~~
```
In this variation, `some_port_name` is driven with a specific constant index in one version, and with a different index in the new version (both exist, different line numbers).

**To Reproduce**
At the moment I have no Verilog file available to reproduce.

**Additional context**
Looking at the changelog, I assume both are correct, and the different selection is due to the changes in `SmallVector` changing some hash algorithm or something. Since the note says "driven here" and not "first driven here" I can't even complain any of the versions is wrong.

MikePopoloski, 2022-10-11:
Thanks for the report. Very likely I'm iterating over an unordered_map somewhere; I'm supposed to sort the results of any such iteration if the result would be user visible but clearly I'm missing that somewhere.

MikePopoloski, 2022-11-13:
I believe I have fixed the source of this in 43082b35b160a36a31ab04add4db53b1e845e5ad

closed 2022-11-13 by MikePopoloski