MikePopoloski/slang #634: Compilation instance destructor triggers SIGABRT

apatern0, 2022-10-11:
**Describe the bug**
When instantiating Compilation, its destructor sometimes fail.

**To Reproduce**
Simply compile and run the following:

```
#include <slang/ast/compilation.h>
int main (void) {
     Compilation c;
}
```

MikePopoloski, 2022-10-11:
This is not enough info to reproduce; there are thousands of tests in the library that exercise code exactly like this; you need to provide more info about how you're building and linking the library into your program.

apatern0, 2022-10-12:
> This is not enough info to reproduce; there are thousands of tests in the library that exercise code exactly like this; you need to provide more info about how you're building and linking the library into your program.

You're right, I extended the message above with more information.

Thanks!

MikePopoloski, 2022-10-12:
It's not enough to just manually link to the library, because if you don't get compilation and linking flags correct you'll have exactly the problem you're experiencing (for example, linking to the debug version of the library requires the `DEBUG` define be set when including slang headers or else you'll get an ABI mismatch.

You might be able to get away with just adding a `-DDEBUG` to your build for now, but to be robust you need to get the full set of flags from cmake by using cmake yourself, or probably something like pkgconfig if you don't want to use cmake.

apatern0, 2022-10-12:
Thanks, using -DDEBUG indeed solves the problem (and others I've been experiencing).

As for CMake, I didn't know it would be necessary, but of course, I can try and implement it. Is there a minimum working example of how such a CMakeLists.txt would look like?

Thanks again!

MikePopoloski, 2022-10-13:
It's on my TODO list to make some examples and write some documentation on integrating the library but haven't done that yet. I'll try to get to it soon though.

closed 2022-10-20 by apatern0