MikePopoloski/slang #641: Duplicate modules (revisited)

udif, 2022-10-19:
**Describe the bug**
What should be the behavior when allowing duplicate modules?

**To Reproduce**
Consider this code:
```
module dbl;
generate
  $info("first!");
endgenerate
initial
  $display("first!");
endmodule

module dbl;
generate
  $info("second!");
endgenerate
initial
  $display("second!");
endmodule

module top;
dbl dbl();
endmodule
```

When running under two commercial tools it seems that the **first** module is the one that gets elaborated.  
When elaborating under slang, the **second** module is elaborated.

Should we change the behavior in #601 and #618?
Should this be an optional flag?

**Additional context**
[https://www.edaplayground.com/x/CUwG](https://www.edaplayground.com/x/CUwG)
Note the use of the `-allow duplicated_units` .
(Use `-Wno-duplicate-definition` with slang).

MikePopoloski, 2022-10-19:
It doesn't matter much to me either way. It's a pretty niche feature, so I'm happy for you to just change the ordering to be whichever way you want it to be.

MikePopoloski, 2023-03-25:
Made this change in 5debcc2acc748e0e6fecb8aeb04122fa221ae663

closed 2023-03-25 by MikePopoloski