MikePopoloski/slang #645: Unnamed scope issue

HungMingWu, 2022-10-31:
**Describe the bug**
I don't know whether it is a bug.
But I see the difference behavior between verilator and slang

Here is the test case
``` verilog
class C;
        task T();
                int arr[5];
                int ccc;
                blk_1: foreach(arr[j]) begin
                        int ccc;
                        ccc = j;
                end
        endtask
endclass
```

In verilator's behavior, the second `ccc` belong to unnamed scope.
But current slang treats them as the same scope.
They have two different symbols, but identical hierarchical path.
I am not sure whether it is right.

MikePopoloski, 2022-10-31:
Can you give more information about what the problem is? The variables are not actually accessibly via hierarchical path since they are in unnamed blocks so I'm not sure what you're observing.

MikePopoloski, 2022-10-31:
Are you sure they're in the same scope? I don't think that's the case, otherwise there would be an error issued about duplicate variable names. There should be separate scopes (instances of StatementBlockSymbol), it's just that those blocks are unnamed. Could you post actual output that shows the problem?

HungMingWu, 2022-11-01:
Sorry, I found some mistake from my implementation.
The issue can be closed until I met other problems.

MikePopoloski, 2022-11-02:
Sorry, I'm still not following your question. Why do j and blk_2 need to be in different scopes?

When I run your example with --ast-json I get the following output describing the structure, which seems ok to me:
```
"members": [
            {
              "name": "T",
              "kind": "Subroutine",
              "addr": 1620316891400,
              "members": [
                {
                  "name": "arr",
                  "kind": "Variable",
                  "addr": 1620316891808,
                  "type": "int$[0:4]",
                  "lifetime": "Automatic"
                },
                {
                  "name": "blk_1",
                  "kind": "StatementBlock",
                  "addr": 1620316891984,
                  "members": [
                    {
                      "name": "j",
                      "kind": "Iterator",
                      "addr": 1620316897576,
                      "type": "int"
                    },
                    {
                      "name": "blk_2",
                      "kind": "StatementBlock",
                      "addr": 1620316892144,
                      "members": [
                        {
                          "name": "k",
                          "kind": "Iterator",
                          "addr": 1620316897944,
                          "type": "int"
                        },
                        {
                          "name": "",
                          "kind": "StatementBlock",
                          "addr": 1620316892304,
                          "members": [
                            {
                              "name": "ccc",
                              "kind": "Variable",
                              "addr": 1620316892464,
                              "type": "int",
                              "lifetime": "Automatic"
                            }
                          ]
                        }
                      ]
                    }
                  ]
                },
...
```

HungMingWu, 2022-11-02:
After discussion with my colleague , I think you are right.
Sorry for this, you can close the issue.

MikePopoloski, 2022-11-02:
Ok, no problem.