MikePopoloski/slang #646: library name collision when upgrade to slang-v2.0

vowstar, 2022-11-02:
**Describe the bug**

Before I used the source code of [v1.0](https://github.com/MikePopoloski/slang/releases/tag/v1.0) to compile and install without problems.

Later, when [v2.0](https://github.com/MikePopoloski/slang/releases/tag/v2.0) was released, I tried to compile and install it with the source code of [v2.0](https://github.com/MikePopoloski/slang/releases/tag/v2.0), and found that two paths were generated that conflicted with the existing ``libslang.so`` of the system:

```bash
 * Detected file collision(s):
 *
 * /usr/lib64/libslang.so
 * /usr/lib64/libslang.so.2
```

**To Reproduce**

This problem is easy to reproduce on CentOS 7. For example, you can install any package that depends on ``slang``(not this repo), such as ``newt``,``libcaca,``nettop``,``util-linux``, etc.

Fastest way to reproduce:

```bash
yum install slang
```

Then CentOS 7 will have:

```bash
/usr/lib64/libslang.so.2
/usr/lib64/libslang.so.2.2.4
```

And then download https://github.com/MikePopoloski/slang/releases/tag/v2.0 source code, compile and install:

```bash
git clone -b v2.0 https://github.com/MikePopoloski/slang.git
mkdir build && cd build && cmake .. && make -j && make install
```
Then the two files will conflict, resulting in corruption.

**Additional context**

Except on CentOS, on other operating systems, library name conflicts will still cause program running errors.

For example debian/ubuntu based OS:

https://packages.debian.org/search?arch=any&mode=filename&searchon=contents&keywords=libslang.so.2

For example:
[AlmaLinux 9](https://pkgs.org/download/libslang.so.2()(64bit)#)
[AlmaLinux 8](https://pkgs.org/download/libslang.so.2()(64bit)#)
[ALT Linux P10](https://pkgs.org/download/libslang.so.2()(64bit)#)
[ALT Linux P9](https://pkgs.org/download/libslang.so.2()(64bit)#)
[ALT Linux Sisyphus](https://pkgs.org/download/libslang.so.2()(64bit)#)
[Amazon Linux 2](https://pkgs.org/download/libslang.so.2()(64bit)#)
[Amazon Linux 1](https://pkgs.org/download/libslang.so.2()(64bit)#)
[CentOS 9 Stream](https://pkgs.org/download/libslang.so.2()(64bit)#)
[CentOS 8 Stream](https://pkgs.org/download/libslang.so.2()(64bit)#)
[CentOS 8](https://pkgs.org/download/libslang.so.2()(64bit)#)
[CentOS 7](https://pkgs.org/download/libslang.so.2()(64bit)#)
[Fedora 36](https://pkgs.org/download/libslang.so.2()(64bit)#)
[Fedora 35](https://pkgs.org/download/libslang.so.2()(64bit)#)
[Fedora Rawhide](https://pkgs.org/download/libslang.so.2()(64bit)#)
[Mageia 8](https://pkgs.org/download/libslang.so.2()(64bit)#)
[Mageia Cauldron](https://pkgs.org/download/libslang.so.2()(64bit)#)
[OpenMandriva Lx 4.3](https://pkgs.org/download/libslang.so.2()(64bit)#)
[OpenMandriva Lx 4.2](https://pkgs.org/download/libslang.so.2()(64bit)#)
[OpenMandriva Rolling](https://pkgs.org/download/libslang.so.2()(64bit)#)
[OpenMandriva Cooker](https://pkgs.org/download/libslang.so.2()(64bit)#)
[openSUSE Leap 15.4](https://pkgs.org/download/libslang.so.2()(64bit)#)
[openSUSE Leap 15.3](https://pkgs.org/download/libslang.so.2()(64bit)#)
[openSUSE Tumbleweed](https://pkgs.org/download/libslang.so.2()(64bit)#)
[PCLinuxOS](https://pkgs.org/download/libslang.so.2()(64bit)#)
[Rocky Linux 9](https://pkgs.org/download/libslang.so.2()(64bit)#)
[Rocky Linux 8](https://pkgs.org/download/libslang.so.2()(64bit)#)

https://pkgs.org/download/libslang.so.2()(64bit)

Since this name has been taken and is well known, it is de facto unavailable in version [v2.0](https://github.com/MikePopoloski/slang/releases/tag/v2.0).
[v1.0](https://github.com/MikePopoloski/slang/releases/tag/v1.0) does not have this name collision problem.

I am trying to make this package for the gentoo distribution, and the test passed in [v1.0](https://github.com/MikePopoloski/slang/releases/tag/v1.0), and have issue in [v2.0](https://github.com/MikePopoloski/slang/releases/tag/v2.0).

MikePopoloski, 2022-11-02:
Hmm, that's unfortunate. I did not know about this "s-lang" library. Not sure what to do about this... a rename would be pretty painful at this point.

vowstar, 2022-11-02:
I temporarily replaced ``libslang.so`` with ``libsvlang.so`` in my local environment. If the final solution is given in upstream, I will follow the upstream way.

vowstar, 2022-11-03:
I work around this issue with static build, but still difficult to package it for linux distribution as a dynamic link library.

vowstar, 2022-11-07:
Thanks for https://github.com/MikePopoloski/slang/pull/650, I'll be using ``libsvlang.so`` for Gentoo Linux packaging.

MikePopoloski, 2022-11-19:
This should be fixed now. The default lib name is set to libsvlang

closed 2022-11-19 by MikePopoloski