MikePopoloski/slang #652: Error when passing interface to instance

MahmoudKMaarouf, 2022-11-08:
**Describe the bug**
A clear and concise description of what the bug is.

These lines of code within a module:
```
an_interface #() u_an_interface ();
some_module #() u_an_instance( .the_inteface (u_an_interface) );
```

causes an error during linting (--lint-only) option

**To Reproduce**
slang --lint-only test.sv
with test.sv including the lines of code above

ERROR;
```
test2.sv:1788:50: error: 'u_an_interface' does not refer to a value
   some_module #() u_an_instance( .the_inteface (u_an_interface) );
```

**Additional context**
It says the interface does not refer to a value. But it has been declared above as an interface.

MahmoudKMaarouf, 2022-11-10:
@MikePopoloski Would you be able to provide an update on this? I would like to use this linter, but this error has been a huge limitation. 
Furthermore, for this linter, I have commented out the line
`{diag::UnknownPackage, std::make_tuple("UnknownPackage"sv, "unknown package '{}'"sv, DiagnosticSeverity::Error, ""sv)},
`in DiagCode.cpp because I did not want the linter to yield unknown package errors. I feel this shouldn't be an error in a linter.

Thank you.

MikePopoloski, 2022-11-11:
I will look into this when I get a chance.

MikePopoloski, 2022-11-13:
Fixed in b92a874301898581a5778f72b5cf19abfa159ec1 and aa81792bc9e2f4ea8e41868eb1ab29fe6aa542ca

closed 2022-11-13 by MikePopoloski