MikePopoloski/slang #654: got error result on ForLoopStatement state

HungMingWu, 2022-11-09:
**Describe the bug**
I try to implement a StatementVisitor

``` cpp
struct StatementVisitor {
    void visit(const ForLoopStatement& stmt) {
        for (const auto& init : stmt.initializers) {
            // do something
        }
    }
};
```
It works well in this case
``` cpp
class C;
        task T();
            int of;
            for (of = 1; of <= 5; of++) begin 
            end
        endtask
endclass
```
But it fail
``` cpp
class C;
        task T();
            for (int of = 1; of <= 5; of++) begin 
            end
        endtask
endclass
```
Due to the node is `ForVariableDeclarationSyntax`, not `ExpressionSyntax`
It cannot process correctly.
``` cpp
Statement& ForLoopStatement::fromSyntax(Compilation& compilation,
                                        const ForLoopStatementSyntax& syntax,
                                        const ASTContext& context, StatementContext& stmtCtx) {
    // ignore
    if (hasVarDecls) {
        // ignore
    }
    else {
        for (auto initializer : syntax.initializers) {
            auto& init = Expression::bind(initializer->as<ExpressionSyntax>(), context,
                                          ASTFlags::AssignmentAllowed);
            initializers.push_back(&init);
            anyBad |= init.bad();
        }
    }
}
```

MikePopoloski, 2022-11-09:
If the loop variables are a declaration then you need to look in the containing scope for a VariableDeclStatement which will have the initializer (all variables need to live in a scope, they can't live in a statement directly).

HungMingWu, 2022-11-09:
Does valuesymbol contain initial value?
I am not sure.

closed 2022-11-10 by MikePopoloski